mik3y/usb-serial-for-android #510: Unable to use USBSerialPort class to open the UsbDeviceConnection connection

mfran89, 2023-07-06:
My goal is to connect a Ardupilot Pixhawk device that uses the FTDI chipset to a Samsung tablet. From reading the readme documentation here and on androids (https://developer.android.com/guide/topics/connectivity/usb/host#java) I am confident that I went through the right steps to get permission granted to interface to the USB device. However when I use the port.open(connection) method I get a IOExpection error "port connection issue java.io.IOException: Reset failed: result=-1".

Here are steps I followed to get to this point
1. Connect USB device to tablet
2. Get a popup from tablet asking if I want to connect to the device
3. Through the usbmanager I run the requestpermission() and pass the pending intent method with the FLAG_MUTABLE flag in it. Current version of sdk is 13. 
4. The broadcast receiver catches this request and grants permission (like the documentation on Android explains) and with in that "if logic" I include the code that the README.md explains to use . This includes the custom prober, driver, and finally the port. It is here that my code gets stuck and cannot get past. 

I can confirm that the USBDeviceConnection is good because I see the getFileDescriptor() returns a value greater than -1. However after I try to use the port.open() call the fileDescriptor() returns a -1. Also the UsbDevice obj will return a device name such as /dev/bus/usb/001/051 which I assume should be a viable path

Looking forward to hearing back , please let me know if I need to give anymore information .

kai-morich, 2023-07-06:
does it work with example app in this repo or https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal where you can configure custom vid/pid mapping?

mfran89, 2023-07-06:
I did not use the example app yet, but I did use the example code that was recommended to connect to the device. Also I did use the serial usb terminal app and found the USB device on there and set the driver to CDC and I saw stream of data coming through granted it was in an unreadable format.

kai-morich, 2023-07-06:
last comment says CDC device, first comment says FTDI. Did you use the wrong driver in your custom prober?

mfran89, 2023-07-06:
Yeah my mistake was using the FTDI driver instead of the CDC one. Once I updated the custom prober to this <code>
customTable.addProduct(usbDevice.getVendorId(), usbDevice.getProductId(), CdcAcmSerialDriver.class);
</code>I was able to fix the issue of opening the connection. The serial usb terminal app helped alot too, thanks so much!

closed 2023-07-06 by mfran89