mik3y/usb-serial-for-android #511: Can I change the targeted compile SDK lower?

amarz2195, 2023-07-07:
Can I change the targetSDKVersion to a lower version? Lower to 28 or so. Because my targeted hardware is running an android version of 9.0

kai-morich, 2023-07-07:
This should not be necessary, as newer version can run on older Android.
If you really want to, you might have to downgrade dependencies like gradle.

amarz2195, 2023-07-09:
apologies for the late reply.
as of now, my target compile sdk is 33 and I try tunning it in android9.0.
the apk can't be installed. I suspect because of the target compile sdk.
but, my minimum sdk version is 17.
so as I browse through the browser, it says that the apk can be installed as long as the minimum sdk is set lower than 28.
can you enlighten me on this matter?
another thing, which dependencies should be downgrade to change the target compile sdk?
thank you in advance

kai-morich, 2023-07-09:
if the minSdk is low enough, it should work. 
Maybe one of your other dependent components has a too high minSdk.
I don't know what to downgrade, you have to try until all errors are gone

amarz2195, 2023-07-11:
thank you sir for the information given

closed 2023-07-11 by amarz2195