mik3y/usb-serial-for-android #513: onRunError:

BlueSky15171, 2023-07-10:
```java
Run ending due to exception: Queueing USB request failed
java.io.IOException: Queueing USB request failed
at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:200)
at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:169)
at com.hoho.android.usbserial.util.SerialInputOutputManager.step(SerialInputOutputManager.java:225)
at com.hoho.android.usbserial.util.SerialInputOutputManager.run(SerialInputOutputManager.java:203)
 at java.lang.Thread.run(Thread.java:764)
```
### I confirmed that I received the message and also replied to it. May I ask how should I solve this problem?

kai-morich, 2023-07-10:
please provide more context, e.g. did you disconnect the device, did it happen immediately after connection or hours later, ...

BlueSky15171, 2023-07-11:
```
--------- beginning of main
--------- beginning of system
2023-07-11 09:43:33.414  1403-4345  ActivityTrigger         system_process                       E  activityResumeTrigger: not whiteListedcom.itskylin.pc/com.itskylin.pc.MainActivity/1
2023-07-11 09:43:33.437  1403-2073  ActivityTrigger         system_process                       E  activityResumeTrigger: not whiteListedcom.itskylin.pc/com.itskylin.pc.MainActivity/1
2023-07-11 09:43:33.487  6549-6549  ProlificSerialDriver    com.itskylin.pc                      D  usbVersion=110, deviceVersion=400, deviceClass=0, packetSize=64 => deviceType=DEVICE_TYPE_HX
2023-07-11 09:43:33.487  6549-6597  SerialInpu...putManager com.itskylin.pc                      W  Run ending due to exception: Queueing USB request failed
                                                                                                    java.io.IOException: Queueing USB request failed
                                                                                                       at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:200)
                                                                                                       at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:169)
                                                                                                       at com.hoho.android.usbserial.util.SerialInputOutputManager.step(SerialInputOutputManager.java:225)
                                                                                                       at com.hoho.android.usbserial.util.SerialInputOutputManager.run(SerialInputOutputManager.java:203)
                                                                                                       at java.lang.Thread.run(Thread.java:764)
2023-07-11 09:43:33.489  6549-6597  SerialListener          com.itskylin.pc                      E  onRunError: 
                                                                                                    java.io.IOException: Queueing USB request failed
                                                                                                       at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:200)
                                                                                                       at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:169)
                                                                                                       at com.hoho.android.usbserial.util.SerialInputOutputManager.step(SerialInputOutputManager.java:225)
                                                                                                       at com.hoho.android.usbserial.util.SerialInputOutputManager.run(SerialInputOutputManager.java:203)
                                                                                                       at java.lang.Thread.run(Thread.java:764)
2023-07-11 09:43:33.490  6549-6597  SerialInpu...putManager com.itskylin.pc                      I  Stopped
2023-07-11 09:43:33.492  6549-6549  UsbRequestJNI           com.itskylin.pc                      D  init
2023-07-11 09:43:33.496  6549-6605  SerialInpu...putManager com.itskylin.pc                      I  Running ...
```
@kai-morich

kai-morich, 2023-07-11:
if you can still exchange messages, you might have opened a new connection that killed the previous?

BlueSky15171, 2023-07-11:
I'll try to disconnect before connecting.

BlueSky15171, 2023-07-11:
I disconnected before connecting, it seems like there was no error reported。
@kai-morich Thanks!

Jai-GAY, 2023-07-31:
@BlueSky15171 

> I disconnected before connecting,

I think I got a similar problem. can you share what disconnect here referring to?  It happens during resource closing for my case.

D  close
E  Error Reading: Attempt to invoke virtual method 'int android.hardware.usb.UsbDeviceConnection.controlTransfer(int, int, int, int, byte[], int, int)' on a null object reference
   Assuming inaccessible USB device.  Closing connection.
   java.lang.NullPointerException: Attempt to invoke virtual method 'int android.hardware.usb.UsbDeviceConnection.controlTransfer(int, int, int, int, byte[], int, int)' on a null object reference
   	at com.hoho.android.usbserial.driver.CommonUsbSerialPort.testConnection(CommonUsbSerialPort.java:162)
   	at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:195)
   	at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:169)
   	at org.droidplanner.services.android.impl.communication.connection.usb.UsbHohoConnection.readDataBlock(UsbHohoConnection.java:221)
   	at org.droidplanner.services.android.impl.communication.connection.usb.UsbConnection.readDataBlock(UsbConnection.java:107)
   	at org.droidplanner.services.android.impl.core.MAVLink.connection.MavLinkConnection$2.run(MavLinkConnection.java:146)
   	at java.lang.Thread.run(Thread.java:1012)
I  Exiting manager thread.
E  Error Sending: Error writing 14 bytes at offset 0 of total 14
   java.io.IOException: Error writing 14 bytes at offset 0 of total 14
   	at com.hoho.android.usbserial.driver.CommonUsbSerialPort.write(CommonUsbSerialPort.java:265)
   	at org.droidplanner.services.android.impl.communication.connection.usb.UsbHohoConnection.sendBuffer(UsbHohoConnection.java:238)
   	at org.droidplanner.services.android.impl.communication.connection.usb.UsbConnection.sendBuffer(UsbConnection.java:116)
   	at org.droidplanner.services.android.impl.core.MAVLink.connection.MavLinkConnection$3.run(MavLinkConnection.java:195)
   	at java.lang.Thread.run(Thread.java:1012)

kai-morich, 2023-08-01:
you very likely closed the connection in a different thread, so read aborts with NPE as on mConnection is null, I will improve the exception message, but when your are in closing phase you better ignore all exceptions.

BlueSky15171, 2023-08-07:
@Jai-GAY I have repeated the connection.

kai-morich, 2023-08-24:
exception improved with 399d3c9c2f241212ce9213c076b5426bd34a81af and 35fdeb1e1355cdddc3b208949a9598cd6f17da8f

BlueSky15171, 2023-08-29:
@Jai-GAY Thanks

closed 2023-08-29 by BlueSky15171