mik3y/usb-serial-for-android #514: CH341A meet  java.io.IOException: Expected 0x0 byte, but get 0xff [init #1]

ethi1989, 2023-07-11:
when i use ch341a , I meet java.io.IOException: Expected 0x0 byte, but get 0xff [init #1],
I have no idea ,is it the hardware error or the library error
vendorId : 6790
productId : 21795

kai-morich, 2023-07-11:
I am not aware of issues with CH34X devices. Is it very new or very old? please try a different device.

ethi1989, 2023-07-12:
> I am not aware of issues with CH34X devices. Is it very new or very old? please try a different device.

ok, I only have a usb relay, I will buy another Ch34X, but may I modify the code like this? does it result anything?
FROM
`		private void initialize() throws IOException {
			checkState("init #1", 0x5f, 0, new int[]{-1 /* 0x27, 0x30 */, 0x00});

			if (controlOut(0xa1, 0, 0) < 0) {
				throw new IOException("Init failed: #2");
			}`
TO
`		private void initialize() throws IOException {
			checkState("init #1", 0x5f, 0, new int[]{-1 /* 0x27, 0x30 */, -1});

			if (controlOut(0xa1, 0, 0) < 0) {
				throw new IOException("Init failed: #2");
			}`

closed 2023-10-18 by kai-morich