mik3y/usb-serial-for-android #516: Having issues using the SerialInputOutputManager class

mfran89, 2023-07-13:
Currently I am trying to use the SerialInputOutputManger class that comes with  implementation 'com.github.mik3y:usb-serial-for-android:3.5.1'. But I am able to call some of the setters that are supposed to be available such as start(), setReadTimeout() and others. So I opened the class file to see why this was the case and realized the class file I have does not match the java file that is on the online repo for this class . Which makes me think that somehow I am pulling the wrong version of the code . I followed the instructions to properly update the build.gradle file to include the repo and I have attached the class code in this message. 

Let me know if I am doing something wrong , look forward to hearing back.
[SerialInputOutputManager.txt](https://github.com/mik3y/usb-serial-for-android/files/12044341/SerialInputOutputManager.txt)

kai-morich, 2023-07-14:
how do you import the library? 
As mentioned in [quick start](https://github.com/mik3y/usb-serial-for-android#quick-start) or did you copy as subfolder into your project?

mfran89, 2023-07-14:
I followed the steps in the quick start 
1.Add jitpack.io repository to your root build.gradle which I assume is the build.gradle in app folder
2. Add library 
```
dependencies {
    implementation 'com.github.mik3y:usb-serial-for-android:3.5.1'
}
```

kai-morich, 2023-07-15:
the decompiled class file is from an older version, maybe something is wrongly cached in Android Studio, gradle, ...
Please try File -> Invalidate Caches, Build -> Clean Project, ... in Android Studio

mfran89, 2023-07-24:
Sorry I thought I responded to this last message. I did do the following steps you mentioned 

1. Invalidate Caches
2. Clean project
3. Build Project

I am still not seeing the solution. For now it's not a big issue for me but I am curious if others have run into this issue. I dont understand how I could have gotten a older version, but I am sure it has something to do with my IDE if no one else has run into this problem.

closed 2023-08-25 by mfran89