mik3y/usb-serial-for-android #519: STM32 bad data

guyber, 2023-07-20:
I am using the lib on Samsung s10, with an STM32 device (Vendor 0483, Product 5740), it seems to work fine, 
but when I am sending data above 1500bytes, I get the data but the last byte (or bytes) seems to be always 0x00, i do receive the same byte count but it seems that the buffer is not fully copied to the result buffer - i tried to send the same buffer to a pc and I received the data OK so it seems like an issue in android (or in the android implementation).
I tried this both with my code and with the Serial USB terminal 1.49 by kai-morich.
the connection is on 460800, 8, n, 1
the problem doesn't appear on smaller messages (1400bytes) or when using CP2104

does this component support the STM32 ?

kai-morich, 2023-07-20:
should work with STM32. Data is split by the library in chunks of USB packet size, usually 64 bytes. I suspect the recieve buffer in your STM32 app is too small.

guyber, 2023-07-20:
Tnx for the quick answer, the thing is that it doesn't work with the
serial  usb terminal as well, and the same code works fine with cp2xxx  on
the same data.

guyber, 2023-07-24:
> the CP2104 is attachad to STM32 serial port, which very likely has a
> different buffer handling

hi,

i think you got it wrong, my embedded software is sending big chunks of
data, the chunks are received OK by a PC application - that probably means
that the sending (writing) app works fine, when i connect the usb to
android device i get the data with the correct data until byte 1466, all
the bytes above this limit are zero - this probably means that for some
reason the received data is not copied to the input buffer in the android -
the same behaviour is observed in both the serial terminal 1.49 app and my
app that uses the *mik3y/usb-serial-for-android.*

kai-morich, 2023-07-24:
No idea, just guessing:
1466 looks like TCP net capacity. USB requests have a limited size, maybe Windows and Android use different size and a buffer that is also shared with networking is overflowing here?

closed 2023-09-14 by kai-morich