mik3y/usb-serial-for-android #524: App keeps rebooting after receiving data or reconnecting after recent connection

yamin8000, 2023-07-26:
My issue may look unprofessional, but somehow, on some devices (Honor/Car Navigation Android devices), this library keeps rebooting the device, Sometimes during data transfer and sometimes after reconnecting after a recent disconnection, On many other devices it's completely OK.
I tried to find the exact Exception but until now I failed to catch an Exception, maybe even it doesn't throw anything at all.  
You may ask for a code sample which I can provide but it's nothing unordinary and I'm using `SerialInputOutputManager.Listener` and `onNewData` listener.  
I tried to catch exceptions and send them to Crashlytics and a custom exception-saving mechanism but until now nothing is thrown. I think saving exceptions to a file inside the device is a better idea than even monitoring the device using Wifi ADB.  
I never expected a simple code could cause an Android device to reboot, does anybody had a similar issue working with USB Serial?  
I suspected that the USB device that sends data maybe is the issue and uses too much power and changed it and also used another [simple app](https://play.google.com/store/apps/details?id=com.csa.serialmonitor&hl=en) to monitor data and it works fine.

kai-morich, 2023-07-29:
Could be to high power consumption or a bug in the Linux kernel of the particular Android device, both unlikely to be caught on application level.

yamin8000, 2023-07-29:
> Could be to high power consumption or a bug in the Linux kernel of the particular Android device, both unlikely to be caught on application level.

It's silly but the problem was fixed for now when the power adapter of that particular Android device was changed.
Thank you.

closed 2023-07-29 by yamin8000