mik3y/usb-serial-for-android #525: Android to Arduino physical device code burn using OTG

ayushimundra30, 2023-08-11:
I went through your repo found useful Android USB host OTG mode enable .

I have query this is just an library to enable the USB host OTG mode in Android latest version for an serial communication . We have to add this library code in existing  project with actually do burning code to physical device Arduino using ATmega328 device driver .
 Link : https://github.com/ksksue/PhysicaloidLibrary
 Currently using above code I am getting connectivity issue . 

Please correct me if my understanding wrong or directly using your code device drivers physical boards inbuild and burn the code ? 

Regards
Ayushi Mundra

kai-morich, 2023-08-12:
You can use both libraries for communication but only the physicaloid or other library for firmware upload. You have to try out if usb-serial-for-android library solves your communication issues. 

If your device does long lasting operations and not handle usb data in the main loop you might get data loss or other communication issues.

ayushimundra30, 2023-08-18:
Hi Kai , 
As per suggestion ,Including the library dependencies and manifest file permission will resolve the connectivity issue cannot open device .

kai-morich, 2023-08-23:
firmware upload is functionality on top of this library. please check physicaloid or other resources.

closed 2023-09-14 by kai-morich