mik3y/usb-serial-for-android #530: onNewData is stopping current run

recser, 2023-08-31:
Hello,
a question regarding any recommendations for implementing the onNewData functionality. Currently due to some processing inside onNewData method (splitting, deduping and so on) the execution of the method takes some small amount the time during which other new events of new data may come which interrupts current run and starts evaluating new data. Is there a proven method to handle this kind of scenarios ? I am currently implementing a queue based approach where onNewData would push new data into the queue, and another thread would read from that queue and process separately the data.
Was wondering if this is the right approach or if there are any other ways of achieving a completion of processing before new data comes in?

kai-morich, 2023-09-02:
if your data should be shown on the UI you could process it asynchronously by the UI mainLooper, e.g. as done [here](https://github.com/mik3y/usb-serial-for-android/blob/34e6d989fdf8d542981c0d1ad9c93027c9be7063/usbSerialExamples/src/main/java/com/hoho/android/usbserial/examples/TerminalFragment.java#L173) and  [here](https://github.com/kai-morich/SimpleUsbTerminal/blob/e0e1313527e45e77cd3b3e80638414bc56c6f6aa/app/src/main/java/de/kai_morich/simple_usb_terminal/SerialService.java#L237). If appended to a `TextView` the UI processing gets slow when already a lot of text is shown, so the second linked variant also takes care of _merging_ received data.

recser, 2023-11-20:
thank you for taking your time to answer, appreciated!

closed 2023-11-20 by recser