mik3y/usb-serial-for-android #533: Anyone facing Issue in android12+ or android 13 with this library?

uniqueram, 2023-09-13:
Everything was working fine until Android 11 but things broke afterward, The application is opening as it should in Android 13 but when the print button is pressed from the scale the application keeps on crashing Also using the external device it is hard to logcat the possible error so I am stuck here what went wrong in android 13 when print button is pressed application crashed.

uniqueram, 2023-09-13:
I have tried  #494 without any success.

uniqueram, 2023-09-13:
application crashes only when the data is sent from the scale.

uniqueram, 2023-09-14:
@kai-morich

kai-morich, 2023-09-14:
please provide callstack and use more specific title

uniqueram, 2023-09-14:
As I am unable to log the problem as the application is not connected to android studio while using with the scale via USB cable so asking anyone facing issue while running this application in android 12+

kai-morich, 2023-09-14:
You can connect adb/android-studio over wifi. Setup sometimes a bit shaky for me, e.g. USB adb connection first, then switch to tcp, but sometimes only working on newly rebooted Android device.

Does the same issue happen with https://github.com/kai-morich/SimpleUsbTerminal?

uniqueram, 2023-09-14:
yes 
I am using this?

uniqueram, 2023-09-14:
I have solved the issue in one place I haven't consider the pendingIntent.flag_Immutable:0; after placing this it works.

closed 2023-09-17 by kai-morich