mik3y/usb-serial-for-android #534: Version 3.6.0 is not accessible at Jitpack

neboskreb, 2023-09-17:
## What is the failure
In Github, my gradle build fails to download the version 3.6.0:
```
FAILURE: Build failed with an exception.

* What went wrong:
Configuration cache state could not be cached: field `elements` of `org.gradle.configurationcache.serialization.codecs.ResolutionBackedFileCollectionSpec` bean found in field `__librarySourceSets__` of task `:hal:mapDebugSourceSetPaths` of type `com.android.build.gradle.tasks.MapSourceSetPathsTask`: error writing value of type 'java.util.ArrayList'
> Could not find usb-serial-for-android-3.6.0.aar (com.github.mik3y:usb-serial-for-android:3.6.0).
  Searched in the following locations:
      https://jitpack.io/com/github/mik3y/usb-serial-for-android/3.6.0/usb-serial-for-android-3.6.0.aar
```

The dependency is included in gradle.build file as follows:
```
    // https://github.com/mik3y/usb-serial-for-android
    implementation 'com.github.mik3y:usb-serial-for-android:3.6.0'
```
The repo is configured in as follows:
```
repositories {
    maven { url 'https://jitpack.io' }
}
```

## Details
Trying to browse `https://jitpack.io/com/github/mik3y/usb-serial-for-android/` reveals that folder `3.6.0` *is* present (next to folder `v3.6.0`):
```
3.5.1/
v3.5.1/
...
v3.6.0/
3.6.0/
``` 
However, browsing folder `3.6.0` gives message "Not found".

[image]

In Android Studio, gradle says it was able to get the .pom file from this folder but not able to download the rest:

[image]

## Additional information
Browsing folder `v3.6.0` (with the leading "v") properly returns all artifacts as expected:
```
usb-serial-for-android-v3.6.0-sources.jar
usb-serial-for-android-v3.6.0.aar
usb-serial-for-android-v3.6.0.module
usb-serial-for-android-v3.6.0.pom
usb-serial-for-android-v3.6.0.pom.md5
usb-serial-for-android-v3.6.0.pom.sha1
```

In Android Studio, gradle also able to download everything needed.

## Known workaround

```
    // https://github.com/mik3y/usb-serial-for-android
    implementation 'com.github.mik3y:usb-serial-for-android:v3.6.0'
```

Unfortunately, this numbering is incompatible with Renovate bot, so you will not be getting notifications of the new versions 

## Possible solution
* Proper publishing of version `3.6.0` to Jitpack
* Alternatively, publishing version `3.6.0` to Maven Central

stevenzhou404, 2023-09-19:
Got same issue on my side.

kai-morich, 2023-09-19:
the [3.6.0](https://jitpack.io/com/github/mik3y/usb-serial-for-android/3.6.0/) folder and content is now available. Maybe you have been the first one accessing the folder without v and it took some time for jitpack to build it.

closed 2023-09-19 by kai-morich