mik3y/usb-serial-for-android #535: only receive 3 bytes per packet

CQHO, 2023-09-17:
i use this library working for cp21xx usb device. only 3 bytes received per packet,but it working well with pc ,or other Usb Serial tool, even the Usb Serial tool was written by this libraray also.

kai-morich, 2023-09-17:
please check https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time

CQHO, 2023-09-18:
thanks for your reply.
I download the latest version 3.6.0, run the demo on the android pad,phone ,the same sing happen.
the code  at TerminalFragment ,read func.

int len = usbSerialPort.read(buffer, 2000);

it will wait no time and return len =3 immediately，i did try the timeout with 0, 500,200,1000,2000.....but it also return len = 3, no time to wait...

How can I make it work properly?

CQHO, 2023-09-18:
my IDE android studio is runing on mac, .... is it required run on windows?
I switch to ftdi, cp21. it's same to receive 3 bytes only.

kai-morich, 2023-09-18:
you have to call `read` again, until all your data is received

CQHO, 2023-09-18:
seems here. I try another usb device, FTDI. I meet the issue at first time. it was disapear when I set the correct BitRate in code.I will change the BitRagte with CP21 and try again.

CQHO, 2023-09-18:
@kai-morich I see the issue. the maxpacketsize  of the CP21 UsbEndPoint is 64,but the FTDI is 512 default. how to modify it ?

kai-morich, 2023-09-20:
typically you do not have to worry about packet size. the chips send full packets or whenever some internal timeout is reached. You have to continue reading until all data is received as mentionede [here](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time)

closed 2023-11-29 by kai-morich