mik3y/usb-serial-for-android #536: SerialInputOutputManager being used by my IDE is incorrect need help

mfran89, 2023-09-20:
So I made a ticket awhile ago in regards to this. I just had a chance to get back to this and from what I can see its the decompiled class . However when I reading error messages in the log cat for this class I can click the line and it will reference me to both the decompiled class  SerialInputOutputManager.class , and inside the class I see this top message
```
// Source code recreated from a .class file by IntelliJ IDEA
// (powered by FernFlower decompiler)
```

Then I see your class SerialInputOutputManager.java . But for some reason I am not sure how to reference your class as opposed to the decompiled one.

Any ideas?

Thanks
Mike

kai-morich, 2023-09-21:
you could try to limit visibility of nested library with [`implementation`](https://developer.android.com/build/dependencies#dependency_configurations) instead of `api`, or try changing the order in `dependencies` section

mfran89, 2023-09-25:
I am currently using implementation to add the library , I tired playing with the exclude command within the implementation and that didnt do the trick either.

closed 2024-08-08 by kai-morich