mik3y/usb-serial-for-android #538: How to properly prepare and serve a "controll endpoint"?

topherbuckley, 2023-09-20:
Hello,

I'm just trying to get a slow basic serial communication working between the rp2040 and an Android phone. If I printf to UART via USB (e.g. the pico-example hello_world example), I can read this printf statement on the Android phone via this library, but I'm not sure how to receive serial data on the rp2040. 

I'm trying to make use of the [dev_lowlevel](https://github.com/raspberrypi/pico-examples/tree/master/usb/device/dev_lowlevel) pico-example, but I'm struggling to get it working with this library. If I build and run it in gdb, I see no errors occurring on the rp2040 side, but on the Android side, using just the default example project in the [usbSerialExamples directory of ](https://github.com/mik3y/usb-serial-for-android) it finds a Cdc driver after adding in the custom vid and pid, but it still fails with "connection failed: No control endpoint" YThe relevant bits of code seem to be [here](https://github.com/mik3y/usb-serial-for-android/blob/80a555a189526eb1cbb48102b9d65980eeecaecb/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/CdcAcmSerialDriver.java#L147) which check for this control endpoint. Looks like the endpoint is supposed to satisfy:

[code]if ((ep.getDirection() == UsbConstants.USB_DIR_IN) && (ep.getType() == UsbConstants.USB_ENDPOINT_XFER_INT))[/code]

and I'm only actually seeing  2 for mControlInterface.getEndpointCount(). Are there supposed to be 3 endpoints or just 2?

Whats especially strange is that the [hello_world](https://raw.githubusercontent.com/raspberrypi/pico-examples/master/hello_world/usb/hello_usb.c) code doesn't seem to explicitly set any of these endpoints, but it somehow works, while the  [dev_lowlevel](https://github.com/raspberrypi/pico-examples/tree/master/usb/device/dev_lowlevel) pico-example appears to modify these end points. I'm wondering if you either can understand easily where in this dev_lowlevel example the endpoint is not being created or served correct, or if that's too much to ask, then if you just have a good example of what these end points should look like in some C code?

kai-morich, 2023-09-22:
[here](https://github.com/mik3y/usb-serial-for-android/blob/master/usbSerialForAndroid/src/test/java/com/hoho/android/usbserial/driver/CdcAcmSerialDriverTest.java) you can find some examples on how many interfaces / endpoints there should be

kai-morich, 2023-09-25:
I guess dev_lowlevel does raw USB communication, not CDC-ACM style

closed 2023-09-25 by kai-morich

topherbuckley, 2023-09-25:
@kai-morich thanks for your replies. Have you done anything with a rp2040 or pico and this lib? Just looking for a simpler example to start with as I'm not familiar with most of the USB stack, and am not really hoping to become an expert just for this one-off project I'm working on atm. I couldn't get the dev_lowlevel to work easily, so I ended up just using the rp2040's UART passthrough to the USB as I can just use `printf` and `get_char`, which has it's drawbacks, so I'll loop back on this if I get time to convert over to a true USB implementation.  Loop me in somewhere else if you end up working on any rp2040 support at some point.