mik3y/usb-serial-for-android #539: SerialOutputManager " Run ending due to exception: null java.nio.BufferOverflowException  "                                                                                               java.nio.BufferOverflowException

mfran89, 2023-09-21:
Currently running an older version of the library due to a required external library that includes it. Running into this issue at run time where I get I this error 

```
Run ending due to exception: null
java.nio.BufferOverflowException
	at java.nio.HeapByteBuffer.put(HeapByteBuffer.java:177)
	at java.nio.ByteBuffer.put(ByteBuffer.java:734)
	at *****
	at *****
	at com.hoho.android.usbserial.util.SerialInputOutputManager.step(SourceFile:168)
	at com.hoho.android.usbserial.util.SerialInputOutputManager.run(SourceFile:142)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:463)
	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
	at java.lang.Thread.run(Thread.java:1012)
```

This happens when I am creating a connection to a USB device **sometimes**. I have been granted permission and can create a UsbSerialPort obj that I pass to and do the following 

```
_serialManager = new SerialInputOutputManager(usbSerialPort);
 _serialManager.setListener(new SerialInputOutputManager.Listener() {
//code to handle onNewData() and onRunError()
}
_executorService.submit(_serialManager); //run on a parallel thread
```
The error I mentioned gets called at onRunError(). When this happens I do the following

```
if (_serialManager != null) {
            _serialManager.setListener(null);
            _serialManager.stop();
        }
        _serialManager = null;
```

I am not sure how to handle it,  and why it's occurring.

kai-morich, 2023-09-21:
without the actual values for `at *****` lines it's hard to tell. 

The only place where `ByteBuffer.put` is used, is `writeAsync`. Maybe you have to set a bigger buffer, as all writes are queued and executed after the next read.

mfran89, 2023-09-21:
it is occurring in this function 
```
_serialManager.setListener(new SerialInputOutputManager.Listener() {
            @Override
            public void onNewData(byte[] data) {
                // Sending Serial data to TCP
                    _tcpManager.writeAsync(data); 
```
This is the tcpmanager function
```
public void writeAsync(byte[] data) {
        synchronized (mWriteBufferLock) {
            mWriteBuffer.put(data); //this is the line of code that is crashing
            }
        }
    }
```

I have a parallel thread that has a similar object to SerialInputOutputManager but instead its a tcp server that grabs the serial data from this obj and passes it to another obj.

The buffer size for the SerialInputOutputManager obj inside the decompiled class is 4096, same with the tcp manager.

What I cant figure out is what the error is implying. Is it saying that the amount of bytes being sent by the SerialInputManager to the tcp manager is too much?

kai-morich, 2023-09-22:
writes are queued and executed after the next read. If you cannot change the buffer size, you could add an additional buffer in front, but with that solution you still can only send 4k per read

mfran89, 2023-09-25:
So what I ended up doing for now is increasing the buffer size of the object handing the onNewData() and for now it looks like it is working. Do you have any edge case protections I can use on top of increasing the size?

closed 2023-11-29 by kai-morich