mik3y/usb-serial-for-android #541: java.io.IOException: ControlTransfer DEVICE_TYPE_HX 0x8 failed: -1

bingo618, 2023-10-07:
device=UsbDevice[mName=/dev/bus/usb/004/005,mVendorId=1659,mProductId=8963,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=Prolific Technology Inc. ,mProductName=USB-Serial Controller D,mSerialNumber=null,mConfigurations=[                                                                                      UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=50,mInterfaces=[                                                                                               UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=255,mSubclass=0,mProtocol=0,mEndpoints=[                                                                                               UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=10,mInterval=1]
UsbEndpoint[mAddress=2,mAttributes=2,mMaxPacketSize=64,mInterval=0]                                                                                       UsbEndpoint[mAddress=131,mAttributes=2,mMaxPacketSize=64,mInterval=0]]]]
ports=[ProlificSerialPort device_name=/dev/bus/usb/004/005 device_id=4005 port_number=0]

implementation 'com.github.mik3y:usb-serial-for-android:3.6.0'

kai-morich, 2023-10-07:
Your device was detected as standard Prolific 2303 (without T... or G... suffix)
There are older variants of this standaerd device type that might not work with this library.
How old is this chip?

bingo618, 2023-10-08:
I don't know how old the chip is, but many times it can be connected successfully, and sometimes it will give an error.

kai-morich, 2023-10-08:
maybe your phone cannot provide enough power. have you tried external power supply?

bingo618, 2023-10-10:
It can run normally after replacing the usb to serial port conversion cable. Thank you!

closed 2023-10-10 by kai-morich