mik3y/usb-serial-for-android #542: USB for debug and for device communication

mpiffari, 2023-10-27:
Hello folks and @kai-morich ,
A simple question: do you know a way to use USB connection both for debug and for connection with an external USB device?

Jai-GAY, 2023-10-27:
[Connect to a device over Wi-Fi](https://developer.android.com/tools/adb#connect-to-a-device-over-wi-fi), this out.

mpiffari, 2023-10-27:
Hi @Jai-GAY,
is this the only solution available? I've tried this in the past weeks, and it seems so slow (with a Xamarin forms app).

Someone has implemented an hw solution (maybe with particular cable connection)?

Jai-GAY, 2023-10-27:
> Someone has implemented an hw solution (maybe with particular cable connection)?

Don't think so. [Power + data, yes](https://shp.ee/d8988at) . Power+ data + data, I haven't see one yet.

kai-morich, 2023-10-27:
With external USB device the port is switched to host mode with OTG adapter. When debugging it is in slave mode with regular USB cable. I really doubt that this can be mixed.

closed 2023-10-27 by kai-morich

mpiffari, 2023-10-27:
@kai-morich thanks for your hints. 
So, how did you approach the problem? Did you use wi-fi debugging?

kai-morich, 2023-10-27:
Yes. Use Wi-Fi debugging