mik3y/usb-serial-for-android #543: Optimizing USB File Transfer Rate with UsbSerialExamples on Linux and Android @cdc

xingxing7290, 2023-11-03:
"Hello, I've been using the UsbSerialExamples sample and have developed with it. The development scenario involves a Linux device emulating a CDC slave device over USB, and connecting to an Android smartphone. The issue I've encountered is that when the Linux device uses open /dev/ttyGS0 and writes data using write, I intend to transfer files with each write operation consisting of 16*1024+50 bytes (following a specific protocol). However, I've noticed that the USB receiver on the Android side only receives 512 bytes at a time, and I have to assemble them into complete data sets. Is the 512-byte limit fixed for each reception, or is there a way to increase it and improve the file reception rate? I appreciate your help in advance."

xingxing7290, 2023-11-03:
At the same time, there may also be partial loss of data. I set the baud rate to 460800 on Android and Linux, but as mentioned in the previous issues, CDC devices are not suitable for any baud rate

kai-morich, 2023-11-03:
see FAQ:
[When reading a message, why don't all bytes arrive at the same time?](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time)
[I observed read data loss with continuous transfer at high baud rates](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-I_observed_codereadcode_data_loss_with_continuous_transfer_at_high_baud_rates)

xingxing7290, 2023-11-16:
In Android, using USB to receive data, the sender sends a large amount of data, and the Android app is designed to receive this data. While receiving data, if the application is suddenly killed, the sender stops sending data accordingly. However, upon reopening the application, it is noticed that the Android receiving application continues to receive the data that was being sent before. Why does this happen? Could it be related to the buffering mechanism? Should the buffer be cleared when entering the background, or should the data in the buffer be ignored when the application is reopened? How can this be implemented? Thank you very much for your response.

kai-morich, 2023-11-16:
pure serial devices will not stop sending, when the app disconnects. With CDC the other side might detect USB disconnect and stop. There is no API to clear buffers. You could try adding some additional flow control like setting/resetting DTR line