mik3y/usb-serial-for-android #545: Received garbage data onNewData callback

TiksGanesh, 2023-11-17:
Hi Team,
I am using this library in my project. Connecting with CH340 type of hardware. 
Connection with hardware and communication is good. But it fails on below scenario

I am using polling mechanism in my application where sending command to hardware with help of SerialInputOutputManager class and writeAsync method.

Failure Scenario
1. onNewData callback I receive garbage mechanism
2. Response time out and it calls onRunError callback
3. Once exception is thrown on onRunError callback, it does not accept any more command continues to throw exception
4. After step 3, I programatically disconnects and re-connect with hardware. 

Suggest me solution, how to avoid above issues?   

Implementation at my end

`override fun onNewData(data: ByteArray?) {
        data?.let {
            val rawData = String(it, StandardCharsets.US_ASCII)
            buildDataFromHardware(rawData)
        }
    }
`

`
private fun buildDataFromHardware(rawData: String) {
        stringBuffer.append(rawData)
        val indexOfRCLFChar = rawData.indexOf('\r')
        isCompleteDataReceived = indexOfRCLFChar > 0
        if (isCompleteDataReceived){
            val output = stringBuffer.toString()
            stringBuffer.delete(0, stringBuffer.length)
            isCompleteDataReceived = false
            listener.onData(output, operationType)
            Logger.d(output)
        }
    }
`

kai-morich, 2023-11-17:
is the baud-rate slightly correct, or slightly different?

TiksGanesh, 2023-11-17:
@kai-morich, Baud rate is suggested by hardware team. 

Addition Information
There is software for windows machine it reads data correctly. It does not throw garbage.

kai-morich, 2023-11-17:
power supply from your Android device could be to week. Please try external power supply

closed 2024-01-16 by kai-morich