mik3y/usb-serial-for-android #546: OTG PD(Power Delivery) issue

siri2677, 2023-11-19:
I'm currently using a USB device connected to my Android device through an OTG cable. However, when I plug USB-C into the PD port on the OTG cable, the USB device is not detected by USB Manager. Do you happen to have any information on this issue?

kai-morich, 2023-11-29:
maybe that port is only for power, not for data

siri2677, 2023-12-07:
I didn't provide enough information in my previous question. To be more specific:
When connecting a USB device to the USB 3.0 port via an OTG cable, it is possible to detect the device using the UsbManager and establish serial communication through your library. However, if a USB-C type charging connector is plugged into the PD port while a USB device is connected to the USB 3.0 port, it becomes impossible to detect the USB device via the UsbManager.

ckdo8008, 2024-01-05:
Tablet information.(version. Etc)

I had the same symptoms with a cheap Chinese device.
In some cases, charging and use at the same time may not be possible due to the USB version or chipset of the tablet.

Jai-GAY, 2024-01-08:
> However, if a USB-C type charging connector is plugged into the PD port while a USB device is connected to the USB 3.0 port, it becomes impossible to detect the USB device via the UsbManager.

anything to do with this, [same issue as this user](https://community.oneplus.com/thread/1369663022062108673)?
It is also better that you provide an image.

ckdo8008, 2024-01-08:
There is no image.

The following products had the same symptoms. Teclast P25

Jai-GAY, 2024-01-08:
> In the case of Teclast P25, it did not work with three types of USB hubs.

P25T, I doubt is even [an OTG-enabled device](https://en.teclast.com/products/p25t-tablet).

siri2677, 2024-01-16:
The device experiencing this issue is the Samsung Galaxy A51 with Android version 12. The issue did not occur during testing on a Samsung Galaxy S9 with Android version 10.

kai-morich, 2024-01-16:
Not a library issue. Different Android devices with different USB chip set have different power supply capabilities. You have to try.

Jai-GAY, 2024-01-16:
> If a USB-C type charging connector is plugged into the PD port while a USB device is connected to the USB 3.0 port, it becomes impossible to detect the USB device via the UsbManager

you may [experience the same as this user](https://community.oneplus.com/thread/1369663022062108673).

ckdo8008, 2024-01-25:
[image]

This is not a problem with the library.
However, it is written for the purpose of sharing the content.

In the Lenovo TB-J607F product, depending on the USB Hub, it may operate as a USB host while charging or may only charge. (Android 12)
Both products worked well on the Samsung SM-T720 product. (Android 11)

The two hubs above used the same chipset.
ID 05e3:0610 Genesys Logic, Inc. 4-port hub

However, products that can only be recharged
Bus 001 Device 003: ID 05e3:0751 Genesys Logic, Inc. microSD Card Reader
Bus 001 Device 127: ID 05e3:0751 Genesys Logic, Inc. microSD Card Reader
Bus 001 Device 126: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 125: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Various functions are included.

If you are experiencing the above problem, my advice is to test a product that only has the USB Hub function. In the case of products that have other functions in addition to the USB Hub, the host function appears to be excluded for reasons of power-related stability.

You must purchase a USB Hub that matches your device. Also, I would like this compatibility information to be shared.

pedroplm, 2025-08-12:
Hi, I experienced a similar issue and can confirm that this is not a problem with the lib. The root cause lies in the hardware and USB host-slave configuration.

In USB communication, there is always a host device (which controls the communication) and a peripheral device (which responds). When using an Android device with USB OTG, the phone or tablet must act as the USB host. However, not all OTG cables or adapters correctly establish this host mode.

Many basic OTG cables are passive and rely on the Android device’s hardware to switch to host mode. If the device or the cable does not properly configure the USB host functionality, communication with USB serial devices will fail. This is why some USB serial adapters or peripherals won’t work reliably.

To solve this, you need an OTG adapter with an integrated USB host controller chip, which actively manages the host role and ensures proper power and data lines.

I developed an app that detects USB payloads, some of which require continuous 24-hour operation. Achieving stable power delivery and reliable data communication in this scenario was challenging without proper host hardware support.

In summary, the issue is not with the Android OS or the app, but with the hardware setup, specifically the lack of a true USB host configuration in the OTG connection. I hope this is still helpful.