mik3y/usb-serial-for-android #547: Connection closed warning in logs

recser, 2023-11-20:
Hello,
i keep getting the bellow warnings in the logs and don't know if this is something normal or I am doing something wrong?
The main idea is that I close a port to change its baud rate (dont know if this is the recommended way or I should just change the parameters to new baud rate without close and open each time) from 19200 to 115200 and the other way around. I do it quite often(to print information on a thermal printer) and getting this warnings make me feel like i could be doing something wrong.
Also recently I've started to experience sudden crashes of my application without any logs or dialog to submit feedback, which occurs primarily during cycles when i am writing on the port.
`11-20 19:31:59.581 12866 12989 W SerialInputOutputManager: Run ending due to exception: Connection closed
11-20 19:31:59.581 12866 12989 W SerialInputOutputManager: java.io.IOException: Connection closed
11-20 19:31:59.581 12866 12989 W SerialInputOutputManager: 	at com.hoho.android.usbserial.driver.CommonUsbSerialPort.testConnection(CommonUsbSerialPort.java:162)
11-20 19:31:59.581 12866 12989 W SerialInputOutputManager: 	at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:213)
11-20 19:31:59.581 12866 12989 W SerialInputOutputManager: 	at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read(CommonUsbSerialPort.java:172)
11-20 19:31:59.581 12866 12989 W SerialInputOutputManager: 	at com.hoho.android.usbserial.util.SerialInputOutputManager.step(SerialInputOutputManager.java:225)
11-20 19:31:59.581 12866 12989 W SerialInputOutputManager: 	at com.hoho.android.usbserial.util.SerialInputOutputManager.run(SerialInputOutputManager.java:203)
11-20 19:31:59.581 12866 12989 W SerialInputOutputManager: 	at java.lang.Thread.run(Thread.java:919)
11-20 19:31:59.581 12866 12989 I SerialInputOutputManager: Stopped
11-20 19:31:59.883 12866 12935 D Ch34xSerialDriver: baud rate=9600, 0x1312=0xb282, 0x0f2c=0x000c
11-20 19:31:59.885 12866 12935 D Ch34xSerialDriver: baud rate=9600, 0x1312=0xb282, 0x0f2c=0x000c
11-20 19:31:59.885 12866 12935 D UsbRequestJNI: init
11-20 19:31:59.885 12866 12935 D Ch34xSerialDriver: baud rate=115200, 0x1312=0xcc83, 0x0f2c=0x0008
`
Is this something that can be avoided( maybe by adding a sleep between write and close to change baudrate ?) or this is something that happens because of other reasons?

Any help or suggestion is appreciated!

Thanks

kai-morich, 2023-11-21:
The blocking `SerialInputOutputManager` wakes up on connection close and prints this warning. This is not an issue

recser, 2023-11-22:
Hi, thanks for the reply. 
From the traces collected i can see that the android kills the application, although it is not clear why? Could it be that the work done reading the serial data is deemed as too 'heavy' and the OS kills the application, since i run it in the background ?

kai-morich, 2023-11-23:
to prevent your app getting stopped, you have to use a foreground service + notification while in background as done [here](https://github.com/kai-morich/SimpleUsbTerminal/blob/6658ea860e5164e4a1df294001fc7a8478aaa895/app/src/main/java/de/kai_morich/simple_usb_terminal/SerialService.java#L169)