mik3y/usb-serial-for-android #548: USB Composite Device. Could not get read & write endpoints

diamond-61, 2023-11-27:
Hello! 
Hello I have "composite"
My device has 5 interfaces. There is a description in the screenshot.

When I call findAllDrivers(UsbManager) and get the driver  ```CdcAcmSerialDriver```.

Next I open the device and try to open the serialPort. ```m_serialPort.open(m_connection);```

I get the message "Could not get read & write endpoints".

I discovered that in your code in the file CdsAsmSerialDriver.jac there is a check on line 187.
```
 if (dataInterfaceCount == mPortNumber) {
     mDataInterface = usbInterface;
}
```

  mPortNumber= 0, when the number of dataInterfaceCount = 0, usbInterface has ID = 1 (#1 on screenshot)
This interface has no endpoints. 
I need an interface with ID = 3 (#4 on screenshot).

I removed these lines from the code and I got my desired interface, but it's wrong. This will not work with any other device.
[image]

How can the library be adapted to work with a composite device?

kai-morich, 2023-11-27:
#1 with 0 endpoints looks strange. Does the device work with Windows?
How many ports are returned by `driver.getPorts()`

diamond-61, 2023-11-28:
driver.getPorts() return 1 port

diamond-61, 2023-11-28:
Add scrennshot with interface #0
[image]

kai-morich, 2023-11-28:
If you are not using usb-serial-for-android library version >= 3.6.0, you have to upgrade.
If you already use it, please send logcat output from CdcAcmSerialDriver  class

diamond-61, 2023-11-28:
```
dependencies {
 implementation 'com.github.mik3y:usb-serial-for-android:3.7.0'
}
```

diamond-61, 2023-11-28:
[image]

diamond-61, 2023-11-28:
[image]

kai-morich, 2023-11-28:
the library tries to detect the related interfaces with IAD, but the first row 12 01 00 ... does not match the [spec table 1-1](https://www.usb.org/sites/default/files/iadclasscode_r10.pdf)

closed 2024-04-16 by kai-morich