mik3y/usb-serial-for-android #549: Echo only in read data

erbl-kamstrup, 2023-11-28:
I use the library to communicate with an optical head plugged in USB with an adaptater
I succed to detect the device with intent filer, create the usb serial port, write and read BUT
The only data I receive is an Echo of the data I just send

Am I missing somethin ? There is the code I use

```
UsbManager manager = (UsbManager) getSystemService(Context.USB_SERVICE);
    UsbSerialProber usbDefaultProber = UsbSerialProber.getDefaultProber();

    HashMap<String, UsbDevice> deviceList = manager.getDeviceList();
    if (deviceList.isEmpty()) {
        Toast.makeText(this, "Aucun phériphérique USB detecté", Toast.LENGTH_LONG).show();
        infos.setText("Aucun phériphérique USB detecté");
    }

    for (UsbDevice device : deviceList.values()) {
        Toast.makeText(this, "Device detecté =" + device.getProductName(), Toast.LENGTH_LONG).show();
        Toast.makeText(this, "Device Vendor =" + device.getVendorId() + " Product Id = " + device.getProductId(), Toast.LENGTH_LONG).show();
        UsbSerialDriver driver = usbDefaultProber.probeDevice(device);
        if (driver == null) {
            ProbeTable customTable = new ProbeTable();
            customTable.addProduct(device.getVendorId(), device.getProductId(), Cp21xxSerialDriver.class);
            UsbSerialProber customProber = new UsbSerialProber(customTable);
            driver = customProber.probeDevice(device);
        }
        if (driver != null) {
            Context c = getApplicationContext();
            PendingIntent pendingIntent;
            Intent notificationIntent = new Intent();
            if (android.os.Build.VERSION.SDK_INT >= Build.VERSION_CODES.S) {
                pendingIntent = PendingIntent.getActivity(c,
                        0, notificationIntent, PendingIntent.FLAG_UPDATE_CURRENT | PendingIntent.FLAG_MUTABLE);

            } else {
                pendingIntent = PendingIntent.getActivity(c,
                        0, notificationIntent, PendingIntent.FLAG_UPDATE_CURRENT);

            }
            while (!manager.hasPermission(device)) {
                manager.requestPermission(device, pendingIntent);
            }
            UsbDeviceConnection connection = manager.openDevice(device);
            if (connection == null) {
                Toast.makeText(this, "Connexion impossible", Toast.LENGTH_LONG).show();
                return;
            }
            infos.setText("Envoi en cours...");
            Thread.sleep(1000);
            Toast.makeText(this, "Envoi en cours...", Toast.LENGTH_LONG).show();
            UsbSerialPort port = driver.getPorts().get(0); // Most devices have just one port (port 0)
            port.open(connection);
            port.setParameters(1200, 8, UsbSerialPort.STOPBITS_2, UsbSerialPort.PARITY_NONE);
            request = new byte[port.getReadEndpoint().getMaxPacketSize()];
            request[0] = hexIntToByte(0x80);
            request[1] = hexIntToByte(0x3F);
            request[2] = hexIntToByte(0x01);           
            byte crcMSByte = hexIntToByte(0x05);
            byte crcLSByte = hexIntToByte(0x8A);
            request[3] = crcMSByte;//CRC MSByte
            request[4] = crcLSByte;//CRC LSByte
            request[5] = hexIntToByte(0x0D);//Stop Byte
            SerialInputOutputManager usbIoManager = new SerialInputOutputManager(port, this);
            usbIoManager.start();
            port.write(request, 0);
            String infosText;
            infosText = "Informations \n";
            infosText += "Data envoyée = " + byteArrayToHexString(request) + "\n";
            infos.setText(infosText);

        } else {
            Toast.makeText(this, "AUCUN Driver disponible", Toast.LENGTH_LONG).show();
        }
    }

    spinner.setVisibility(View.GONE);
    } catch (Exception e) {
        String erreur = e.toString();
        Toast.makeText(this, "Erreur = " + erreur, Toast.LENGTH_LONG).show();
        String erreurText;
        erreurText = "ERREUR \n";
        erreurText += "Data envoyée = " + byteArrayToHexString(request) + "\n";
        erreurText += "Erreur Message = " + erreur + "\n";
        infos.setText(erreurText);

    }
```
`@Override
public void onNewData(byte[] data) {
    runOnUiThread(() -> { infos.append(byteArrayToHexString(data)); });
}`

kai-morich, 2023-11-28:
maybe there is a shortcut between TX and RX lines, so you receive what you send

erbl-kamstrup, 2023-11-29:
> maybe there is a shortcut between TX and RX lines, so you receive what you send

Sorry I didn't uunderstand , can you develop please ?

erbl-kamstrup, 2023-12-01:
after some tries , I have in my onNewData the echo THEN the data I expected in the same byte
I will manage to read and use it, thanks for your answer

closed 2023-12-05 by erbl-kamstrup