mik3y/usb-serial-for-android #550: Read the prompt

tonnqueiroz, 2023-11-28:
Is it possible to read the returned prompt before or after sending each command?

kai-morich, 2023-11-29:
Not clear to me, what you are looking for. You can use direct read or event driven read as mentioned in [readme](https://github.com/mik3y/usb-serial-for-android)

tonnqueiroz, 2023-11-29:
@kai-morich I need to automate the sending of commands to routers from several different manufacturers. Below I illustrate the prompts for routers manufactured by HUAWEI. Based on each prompt, my automation should send a different command, so I need to be able to read the prompt!

New Username:
Password:
Confirm password:
Username:
Password:
<Huawei>
[Huawei]

kai-morich, 2023-11-29:
read the response with direct read or event driven read as mentioned in [readme](https://github.com/mik3y/usb-serial-for-android), then send different next command

Uditk2002, 2023-12-21:
Hi @kai-morich @tonnqueiroz  direct read.. is mentioned inside the read() method, if I am right. Even if I call read() as soon as I write it reads 0 bytes. I tried adding delay but still, each and every command that is sent one at a time is recieved or may be appended (late) in the textbox after everything is sent!! . I have the same goal as @tonnqueiroz does.
I am a newbie to android programming. Your help would really mean a lot.
Thankyou

kai-morich, 2023-12-21:
see https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time

closed 2024-01-16 by kai-morich