mik3y/usb-serial-for-android #552: How to return complete data while read a message ?

HarizahSyawal, 2023-12-12:
Hello, i have do some integration from vending machine using mdb protocol and i'm using this library to communicated with. My issue here is whenever i read response from vending machine, it return line by line and the message not completed read in one line.
[image]

it supposed to be like this : 
[image]

Here is my code : 

` private void receivedMessage(ArrayDeque<byte[]> datas) {

        char STX = '\u0002'; // Start of Text
        char ETX = '\u0003'; // End of Text
        SpannableStringBuilder spn = new SpannableStringBuilder();

        for (byte[] data : datas) {
            String msg = new String(data);
            String messageWithoutSTXETX = msg.replace(STX, ' ').replace(ETX, ' ');
            spn.append(messageWithoutSTXETX);
        }
        mTvReceiveText.append(TextUtil.newline_crlf+spn);
        Log.d(TAG,"FINAL MESSAGE "+spn);

        if (String.valueOf(spn).contains("1300") || String.valueOf(spn).startsWith("1300")) {
            handleReceivedMessage(String.valueOf(spn).trim());
        }
        else {
            Log.d(TAG, "not indicate a Vend Request : ");
        }
    }`

kai-morich, 2023-12-12:
see https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-When_reading_a_message_why_dont_all_bytes_arrive_at_the_same_time

HarizahSyawal, 2023-12-12:
it mean,  i should limit the bytes follow my protocol specs right

HarizahSyawal, 2023-12-13:
just now i have try your suggestion based on wiki that you share. The issue is whenever i put breakpoint, let say on the loop while read a messages it return completely but when i run without debug or get breakpoint the messages come out separately.

kai-morich, 2023-12-16:
with debugging you change timing and therefore buffering behavior. Better use logcat output for analysis

closed 2023-12-16 by kai-morich