mik3y/usb-serial-for-android #555: Read ONE line then close connection

NeedNot, 2023-12-19:
I'm trying to simply test a serialPort to check if it's outputting the desired data (there are different ports for different data on this device)

However, I'm not having luck making it work. If I test port 1 it's good, then if I test port 2 it will get data from port 1? but then it's reset so if I test port 2 again it's the correct data. this is repeated for everything. first time it works, second time outputs from the same device as before, 3rd time is back to the first time behavior.

I played around with my code a bit and now it only outputs from the same port

```java
    private static IntByteArrayPair testReceiver(UsbManager manager, UsbSerialPort serialPort) {
        if (manager == null || serialPort == null) return null;
        UsbDeviceConnection connection = manager.openDevice(serialPort.getDevice());
        byte[] response = new byte[serialPort.getReadEndpoint().getMaxPacketSize()];
        if (connection == null) {
            return null;
        }
        try {
            serialPort.open(connection);
            serialPort.setParameters(115200, 8, UsbSerialPort.STOPBITS_1, UsbSerialPort.PARITY_NONE);
            int len = serialPort.read(response, 1000);
            System.out.println(utils.convertByteToHexadecimal(response));
            return new IntByteArrayPair(len, response);

        } catch (IOException e) {
            try {
                serialPort.close();
            } catch (IOException ex) {}
            return null;
        }
        finally {
            try {
                serialPort.close();
            } catch (IOException e) {
            }
        }
    }
```

If i had to guess perhaps the connection isn't closing?
```
2023-12-18 20:21:33.649 13005-13077 UsbDeviceConnectionJNI  net.neednot.planterline              D  close
2023-12-18 20:21:33.745 13005-13016 System                  net.neednot.planterline              W  A resource failed to call close. 
2023-12-18 20:21:33.745 13005-13016 UsbRequestJNI           net.neednot.planterline              D  close
```
not  sure if this is related but i get that everytime i try to run the code

Also it appears like i should be using custom listeners so that when I read the data is new.
any idea how to read ONLY the newest COMPLETE line? this is for something else but i'll need to read the data in realtime

kai-morich, 2023-12-19:
opening multiple connections is explained [here](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-Can_I_open_multiple_connections_in_one_application)

the library does not know, what a 'line' is, see [here](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#when-reading-a-message-why-dont-all-bytes-arrive-at-the-same-time). you can only read all newly available data

NeedNot, 2023-12-19:
> opening multiple connections is explained [here](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-Can_I_open_multiple_connections_in_one_application)

I'm not trying to connect to the same device at once, When I call the function it should open the port that was passed, read a line then close the connection. I'm not sure if the connection is being closed but what I do know is opening a different port doesn't read from that new port.

NeedNot, 2023-12-19:
[image]
here is what I mean, the first time you click on something it shows an output from the last port that was connected, click it again then it shows something from the last device, sooo....

if you had 3 ports
click on port 1 get data from port 1,
click on port 2 get data from port 1,
click on port 3 get data from port 2,
click on port 3 again get data from port 3,
click on any other port and get data from port 3,
click on any port and get data from port the one you just clicked

NeedNot, 2023-12-19:
please delete this I am an idiot. I fixed it

closed 2023-12-19 by kai-morich