mik3y/usb-serial-for-android #559: How can I create a .jar file?

AlbertBence, 2024-01-22:
"If you need a jar file (e.g. for cordova) run gradle task createFullJarRelease and use full.jar"
This is the only description I find here.

It's my first time working on an Android project. I don't even know what a gradle task is, let alone how to run it. I tried making a .jar file using windows cmd java compiler, but it didn't work.

I know I'm probably missing the obvious, but please, someone explain me. I wasted two days of my life trying to create one.

closed 2024-01-24 by AlbertBence

Cesar0112, 2024-10-05:
Please can you give me the step by step guide on how you did it?

kai-morich, 2024-10-05:
see https://github.com/mik3y/usb-serial-for-android/wiki/Build-Variants#jar