mik3y/usb-serial-for-android #560: Host and Accessory communication(NOT AN ISSUE)

HafizMSaad, 2024-02-01:
I'm encountering difficulties establishing communication between two Android devices, where one is designated as the host and the other as the accessory. I attempted to utilize the usb-serial-for-android library on one device and connected the two devices using a wired connection. However, I'm facing issues in establishing a connection and facilitating communication between the devices.

Jai-GAY, 2024-02-02:
Does it sound like a USB hub in the Host machine? if I understand correctly, USB serial can only be one-to-one communication.
So, let say you buy a USB hub and connect to multiple slave devices and the Hub connect to the USB Android Host phone.

something [like that I am buying](https://www.aliexpress.com/item/1005005938817267.html).

HafizMSaad, 2024-02-02:
@Jai-GAY thanks for the prompt response. Actually I want to facilitate communication between 2 android phones, via wired channel using [Android USB framework](https://developer.android.com/develop/connectivity/usb). Phone A will act as the host, and Phone B will act as the accessory. And I would like to share some kind of data, text/image from A to B and B to A.

Jai-GAY, 2024-02-02:
> I want to facilitate communication between 2 android phones, via USB-C cable using [Android USB framework](https://developer.android.com/develop/connectivity/usb)

Have you come to [the wrong place to find an answer](https://stackoverflow.com/questions/34782464/transfer-data-between-two-android-phones-using-usb-host-api)?

By the way, is your connecting cable a PD or OTG cable?

Why you do not consider Bluetooth, wifi-direct, Google Drive, Onedrive, etc?

kai-morich, 2024-02-02:
I would expect that it should work if you use an OTG adapter, as this will switch the Android device in host mode. With USB-C to USB-C cable it unlikely will work

Jai-GAY, 2024-02-02:
> With USB-C to USB-C cable it unlikely will work

ya, in the market seems like all are PD cable only.

HafizMSaad, 2024-02-02:
@Jai-GAY 
Thanks again, but I'm unfamiliar with terms like OTG and PD. While I've come across blogs(related to USB Host and Accessory) discussing OTG, I'm currently using a simple USB-C to USB-C cable to connect my two Android phones. I'm sticking to a wired scenario and not considering alternatives like Wi-Fi Direct due to specific requirements.

HafizMSaad, 2024-02-02:
@Jai-GAY I'm grateful for your help, but the Stack Overflow link you provided doesn't completely address my concerns, even though I'm dealing with the exact same scenario.

kai-morich, 2024-02-02:
you should try with combination of:
- USB-C to USB-A cable
- USB-A to USB-C OTG adapter

HafizMSaad, 2024-02-02:
@kai-morich Thank you. I'll share the results here after testing. 
Just to clarify, I want to make sure: Is it definitely possible to establish communication between two Android phones using a wired channel? I've been struggling with this for the past week and need confirmation.

ckdo8008, 2024-02-02:
If FTDI USB NMC works on Android, you can communicate directly by treating it like a serial connection.
Communication between smartphones is possible by developing an app.
https://ftdichip.com/products/usb-nmc-2-5m/
Data Rates: 3MBaud
 FTDI Internal IC: FT232R

ckdo8008, 2024-02-02:
If you want to use Android Open Accessory, you need to emulate USB Accessory, but it seems to be close to impossible due to Android security.
If you connect two smartphones through a USB Accessory, it will provide similar speeds to the USB NMC above.

HafizMSaad, 2024-02-02:
@ckdo8008 thanks for your input, I have develop an App where I have 2 android phones connected via wire(USB-C to USB-C cable). Connection is established, but when I try to send data from 1 phone to other, the connection gets loss. And I am facing the same issue while using **usb-serial-for-android**.

ckdo8008, 2024-02-05:
I'm sorry. I tested it. It communicates via OTG.

It seems that you need to communicate using controlTransfer or bulkTransfer in file transfer mode. The current library is a serial communication library, so it doesn't seem to be usable. (There is no information on file transfer modes, etc. for compatible devices.)

ckdo8008, 2024-02-05:
If you use developer mode, various approaches are possible by using libraries that implement ADB.
https://github.com/MuntashirAkon/libadb-android

HafizMSaad, 2024-02-05:
@ckdo8008 thanks for sharing your findings/results here. Can you plz tell did you test it between 2 android phones?

ckdo8008, 2024-02-06:
First of all, I am developing on Android 11 or higher. SDK uses 34.
Direct control of OTG failed.

It seems difficult to bypass Android's MTP service in the MTP method.
If you want to communicate using the MTP method, you must create a specific file through MTP and communicate by monitoring the specific file on the other side and reading it if there are any changes. (IPC File)
Create a specific file containing the content to be transferred. Likewise, two-way operation is possible if the other side monitors a specific file using MTP.
If you want to transfer a file from an MTP host, you can just deliver it via MTP.
If you want to transfer a file from an MTP slave, you can enter the file location and transfer flag you want to transfer in a specific file, read the file from the MTP host, and save it on the host side.

If you do not like the IPC communication method using files, you can set it to Midi and process send and receive through MidiReceiver.