mik3y/usb-serial-for-android #561: Jitpack is not serving '3.7.0'

simonegiacomelli, 2024-02-21:
I was hesitant to post this issue because I don't know where to start making sense of it. 
BUT, this is what it looks like is happening to me:

My Android build is failing because ` https://jitpack.io/com/github/mik3y/usb-serial-for-android/3.7.0/usb-serial-for-android-3.7.0.pom`  is not found.

I _don't think_ I broke my build... I think it is reasonable to expect that the above url should exists. 

Let me give you some evidence: if we open the page `https://jitpack.io/com/github/mik3y/usb-serial-for-android/` it gives the list of the available entries; I will paste here a few of them:
```
[...]
v3.6.0/
3.6.0/
master-80a555a189-1/
v3.7.0/
3.7.0/
master-880c0070cb-1/
[...]
```

As you can see you both have `3.7.0` and `v.3.7.0`.
But when we try to access the above url concatenating ['3.7.0'](https://jitpack.io/com/github/mik3y/usb-serial-for-android/3.7.0/) it gives `Not found`.
At least at the time of writing.

It looks like the folder '3.7.0' is not accessible but it is enumerable by the containing folder. 
And it is not the only one though, e.g., '3.6.0' is not accessible. 
Folder '3.5.0' is perfectly fine.

Is Jitpack unable to serve some folder here and there?

joseph-lewis, 2024-02-22:
Same issue here.. I'm going to have to add the library locally in the meantime.. Hoping for a fast fix for this simple issue. This worked for me 5 days ago (The package was last editted 4 days ago)

kai-morich, 2024-02-22:
This package exists with and without v prefix, but jitpack only builds the v variant by default. I triggered the others manually.

closed 2024-02-22 by kai-morich