mik3y/usb-serial-for-android #562: cannot check the connected device.(not a library problem)

ckdo8008, 2024-02-28:
HW: SM-T720 & SM-T725
OS: android 11

Using a product that handles power supply and USB connection (USB hub or Y-type cable)

If you connect the USB while charging for more than a certain period of time, the device may sometimes be unrecognized.
If you reconnect the USB several times, the USB device is not recognized.

UsbSerialProber.getDefaultProber().findAllDrivers(usbManager)
Even if you call all USB devices, you cannot check the connected device.

Unrecognition can be resolved by disconnecting and reconnecting the power supply.
Device recognized.

Until recently, I thought it was a USB hub problem.
The same symptom occurred even when connected with a Y-type cable that does not use the USB Hub chipset.

PD charging does not seem to be good for USB use.
So, it seems that some Android devices prevent charging and USB use together.

Jai-GAY, 2024-02-28:
> If you connect the USB while charging for more than a certain period of time, the device may sometimes be unrecognized.

Do you think your phone is fully charged and intelligently "inform" charger stop supplying as a result the OTG hub power down?

ckdo8008, 2024-02-28:
> > If you connect the USB while charging for more than a certain period of time, the device may sometimes be unrecognized.
> 
> Do you think your phone is fully charged and intelligently "inform" charger stop supplying as a result the OTG hub power down?

I believe that if overheating occurs due to the internal PD USB charging method, the USB connection will not work.

ckdo8008, 2024-03-12:
We are utilizing RS485-USB converters connected to tablets for monitoring devices. The system operates by sending requests from the tablets to each device, with each device responding back. Continuous monitoring is required, so the tablets are powered and charged via USB-C, which is also used to connect the RS485-USB converter. All tablets are standardized on ANDROID 11 OS.

We have about 50 devices in operation, and generally, they function well. However, occasionally, some devices experience app crashes, and attempts to relaunch the app reveal that the RS485-USB converter cannot be found. The solutions to this issue involve either reconnecting the USB-C cable (though sometimes this does not resolve the connection issue) or power cycling the external power supply, which typically resolves the problem.

Problem 1:
We are unable to collect data on why the app crashes. Even Firebase's Crashlytics does not capture the incidents. It seems to be an exception that cannot be handled.

Problem 2:
To reconnect the RS485-USB converter, it seems we require a watchdog mechanism that can short the external power line using a relay component.

We suspect that many tablet or smartphone vendors restrict OTG functionality while charging via USB-C. There are concerns that simultaneous charging and usage might have potential issues, which could be why there are restrictions.

If anyone has encountered similar issues or has found a workaround, could you please share your experiences or solutions?

Useems, 2025-11-07:
Any update? Same here