mik3y/usb-serial-for-android #570: USB Transfer size on FTDI Driver

Covarians, 2024-04-16:
Hello,
I would like to modify the USB Transfer size on the FTDI driver.
Would you know the params to use in the mConnection.controlTransfer() command.

Regards.

FG

kai-morich, 2024-04-16:
all known parameters are in the source code, but I doubt you can change the size

Covarians, 2024-04-16:
I am using the library at 3MBbs on an android phone connected to device with FT430X, using XMODEM (1k Packets) and some time I stop receiving the 62 bytes packet in the middle of a 1024 byte XMODEM packet reception. This is why I would like to increase the packet size.

Covarians, 2024-04-17:
Did you consider using the FTD2XX for android driver in the usb-serial-for-android instead of writing directly to the endpoint0 ?

kai-morich, 2024-04-17:
you could try D2XX directly, but this commercial library will not be integrated into this open source project

closed 2024-04-24 by kai-morich