mik3y/usb-serial-for-android #572: New activity is created when serial USB is connected to a phone

achjaj, 2024-04-22:
I'm trying to add the new USB device detection feature to my app, but a new activity is created instead of updating the running activity. I added the intent filter to my manifest and also copied the required XML file to `res/xml`. 

The relevant part of my Manifest.xml:
```xml
<activity
    android:name=".MainActivity"
    android:exported="true"
    android:label="@string/app_name"
    android:theme="@style/Theme.SerialSensors.NoActionBar">
    <intent-filter>
        <action android:name="android.intent.action.MAIN"/>
        <category android:name="android.intent.category.LAUNCHER"/>
     </intent-filter>
     <intent-filter>
          <action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" />
     </intent-filter>

     <meta-data
         android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
         android:resource="@xml/device_filter" />
</activity>
```

This is how I overridden the `onNewIntent` function, but it is never called (no log or Toast output):
```java
@Override
protected void onNewIntent(Intent intent) {
    super.onNewIntent(intent);
    Log.i("IntentAction", intent.getAction());
        
    if (intent.getAction().equals("android.hardware.usb.action.USB_DEVICE_ATTACHED")) {
        Toast.makeText(this, "New device", Toast.LENGTH_LONG).show();
    }
}
```

Compile SDK is set to 34, and my phone is running Android 12

kai-morich, 2024-04-24:
you need `android:launchMode="singleTask"`, see https://github.com/kai-morich/SimpleUsbTerminal/blob/2ae8796fea3ac8ec1fccd09d3eedb07438ec5344/app/src/main/AndroidManifest.xml#L22

achjaj, 2024-04-24:
Ok, sorry it was my mistake. I put the `android:launchMode` under `<application>` instead of `<activity>`

closed 2024-04-24 by achjaj