mik3y/usb-serial-for-android #574: Communication ISSUE

nadeem-agva, 2024-04-29:
I am using two USBs in my device and two different SerialInputOutputManager Classes one of the USBs stops working randomly and no data is received on that and when I restart the application it starts working. USB 1 is always working and USB2 is not working randomly, it stops reading data at any point in time.

kai-morich, 2024-04-29:
maybe power supplied from android is to weak for the second device

nadeem-agva, 2024-04-30:
but this is only working after rebooting the android device else it is not working even when the app is destroyed and again re created.

nadeem-agva, 2024-04-30:
is there any way possible to restart the communication of that USB port.

kai-morich, 2024-05-01:
please check https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#user-content-Can_I_open_multiple_connections_in_one_application for multiple connections

nadeem-agva, 2024-07-15:
yeah i do checked this but still one of the USB went non-responsive applied all the tactics you provided but still the same issue in one of the USB's