mik3y/usb-serial-for-android #579: Export app crash restart

0cococ, 2024-06-04:
How can I stop communicating with the device? I try to change the device status or unplug the device, which causes the app to crash and restart. I try to close it with mSerialIoManager.stop()
port.close()
connection.close(), but it still crashes. Is there any way to close the device connection as long as the device status changes or is unplugged? I don't want the app to crash and restart.
[image]

kai-morich, 2024-06-04:
that's not a crash, only a warning trace showing the SerialInputOutputManager thread termination reason

0cococ, 2024-06-04:
> 这不是崩溃，只是显示 SerialInputOutputManager 线程终止原因的警告跟踪
This problem caused me to re-run the app  It will search for the device again. It will not cause the app to close directly, but it will re-execute some processes of MainActivity.
[image]

kai-morich, 2024-06-04:
To stop communication yourself,  close the UsbSerialPort.
To stop communication on external error (e.g. disconnect) handle SerialInputOutpManager.Listener.onRunError() and close the port

IagonaDevTeam, 2024-06-14:
Hey @kai-morich, almost created a new issue before checking this one, which is the exact same one I encounter. I tried many (Not to say dozen) of code check in order to avoid this issue without any success. I've got the same warning as @0cococ and after that the onPause of my app is called and everything seems "restarted" but in a very stange way. So I admit this come from the plugin as it never appear if we don't use COM object. I try you solution, so, I added my closeConnection function inside the onRunError, but it doesn't change anything. I've checked the code of the plugin another time (Already done a PR on this project a year before) and think it can be workarounded by adding a Looper on the listerner call. Do you think its viable?

Here is what I do for closing connection:

```
      this.usbIoManager?.listener = null
      this.usbIoManager?.stop()

      try
      {
          this.port?.close()
      }
      catch (e: Exception)
      {
          Log.error("Unable to close port: ${e.message}")
      }

      try
      {
          this.myConnection?.close()
      }
      catch (e: Exception)
      {
          Log.error("Unable to close connection: ${e.message}")
      }
```

Here's the workaround

```
@Override
public void run() {
    synchronized (this) {
        if (getState() != State.STOPPED) {
            throw new IllegalStateException("Already running");
        }
        mState = State.RUNNING;
    }
    Log.i(TAG, "Running ...");
    try {
        if(mThreadPriority != Process.THREAD_PRIORITY_DEFAULT)
            Process.setThreadPriority(mThreadPriority);
        while (true) {
            if (getState() != State.RUNNING) {
                Log.i(TAG, "Stopping mState=" + getState());
                break;
            }
            step();
        }
    } catch (Exception e) {
        if (mSerialPort.isOpen()) {
            Log.w(TAG, "Run ending due to exception: " + e.getMessage(), e);
        } else {
            Log.i(TAG, "Socket closed");
        }
        final Listener listener = getListener();
        if (listener != null) {
            // Here, handle the exception without triggering onPause
            new Handler(Looper.getMainLooper()).post(() -> listener.onRunError(e));
        }
    } finally {
        synchronized (this) {
            mState = State.STOPPED;
            Log.i(TAG, "Stopped");
        }
    }
}
```

Thanks in advance

kai-morich, 2024-06-16:
The listener is invoked in the same background thread as the `SerialInputOutputManager` is running. 
If you want to perform UI operations, you have to switch to the main thread with 'looper.post'. 
This should be done in the Listener, as done in the [example app](https://github.com/mik3y/usb-serial-for-android/blob/275590027b6fbdeffe3ff9293b49594a56c53418/usbSerialExamples/src/main/java/com/hoho/android/usbserial/examples/TerminalFragment.java#L189).

ExPl0siF, 2024-06-16:
> The listener is invoked in the same background thread as the `SerialInputOutputManager` is running. If you want to perform UI operations, you have to switch to the main thread with 'looper.post'. [...]

Thanks for the infos (Sorry, wrong account on the first post). But I don't do any UI operations (Except if you count UsbSerial disconnection and other disconnect things as UI Operations). I use this plugin in order to communicate with custom Serial object and it just get status every 30 seconds and send it to a server online. If I don't disconnect anything in case of failure, do the app can "crash" too? Thanks in advance

IagonaDevTeam, 2024-06-17:
So, I just tested what you've said (Added Handler(Looper.getMainLooper()).post() inside the onRunError) and It didn't change anything, the behavior is still the same. Do you have any other potential fix for this issue?

kai-morich, 2024-06-17:
> Thanks for the infos (Sorry, wrong account on the first post). But I don't do any UI operations (Except if you count UsbSerial disconnection and other disconnect things as UI Operations). I use this [...]

what kind of "crash" do you get? Can you provide a callstack or logcat output?

kai-morich, 2024-06-17:
Android lifecycle is complex. I recommend https://github.com/xxv/android-lifecycle.
To prevent multiple instances of your app, you could try different launchMode as used [here](https://github.com/kai-morich/SimpleUsbTerminal/blob/156a453331663fb4cf365bfe1cb4809a1f960705/app/src/main/AndroidManifest.xml#L16).

IagonaDevTeam, 2024-06-18:
> Android lifecycle is complex. I recommend https://github.com/xxv/android-lifecycle. To prevent multiple instances of your app, you could try different launchMode as used [...]

Thanks, I know Android Lifecycle is complex, but it shouldn't be triggered with the plugin and a warning. I already tried to setup singleTask and singleTop but it just a bad workaround that create other problems. This behavior only happen with USB plugged in and the plugin running. So, do you have more infos about this behavior and maybe a fix? As there is already 3 issues opened that seems to trigger this problem :/ Thank you very much in advance

IagonaDevTeam, 2024-06-24:
Ok! I have some (good?) news about this! I think I've managed to find the root cause of all the problems! The problem was an app restart after the get_status error. The fact is, when this issue is triggered it seems that the COM device (I admitted this come from the device) create a micro-disconnection that trigger the warning (Which is right, as the connection is/seems lost) and the micro-disconnection trigger the Intent registered with the app. SO, the app onPause is called in order to let another instance (Of the same app/Activity) to be launched and this is why there is this strange behavior. So, for all the other that will read this, you have mainly two solutions:

- Kill/Quit the app inside the onPause when get_status error is triggered and let the intent do the thing and restart the app gracefully
- Maybe remove the intent in order for the app/Activty to not be relaunched (But when I remove the intent, it seems that I can't connect anymore to the COM device, but need more test on this point, so, maybe it's possible)

kai-morich, 2024-06-25:
what kind of intent are you using?

IagonaDevTeam, 2024-06-25:
> what kind of intent are you using?

I'm using android.hardware.usb.action.USB_DEVICE_ATTACHED, so when a device is attached it launch my Launchscreen, this is why I'm thinking of micro-disconnection from the COM device

kai-morich, 2024-06-25:
android.hardware.usb.action.USB_DEVICE_ATTACHED is not really needed. Access permissions are handled differently.  USB_DEVICE_ATTACHED is more like a convenience feature for your App and for users. If multiple Apps can handle USB devices the previously selected can be automatically started. 
You should use https://github.com/mik3y/usb-serial-for-android/blob/275590027b6fbdeffe3ff9293b49594a56c53418/usbSerialExamples/src/main/AndroidManifest.xml#L17 to get the intent delivered to the existing App instance.