mik3y/usb-serial-for-android #580: OTG cable unnecessary with type-C?

jactor2, 2024-06-05:
Hi, I've created an app with your library, working perfectly. However, it came to my eye during my research that type-c supports OTG without the need of a OTG adapter, mass-storage devices, controllers, keyboards, etc. no longer requires an otg cable for phones with OTG support and type-c.

I tried connecting without a OTG cord to a mass storage device, and works, yet couldn't succeed when connecting to a serial devices (Cp2102).

Is it possible that this library can connect to CP2102 without an OTG cord and solely rely on Type-C to Type-C Cord? 

Thanks!

kai-morich, 2024-06-06:
By default your Android device acts as USB peripheral. With the OTG adapter it is switched to USB host mode. As summarized in [this paper](https://www.ti.com/lit/wp/slly017/slly017.pdf) these modes can be negotiated with USB-C. I don't know if your Android device support this negotiation, but your CP2102 very likely does not and only connects the USB 2.0 pins of the USB-C connector, so still needs an OTG adapter.

jactor2, 2024-06-07:
Thank you for the advice. It looks like the OTG adapter is unneeded, and the issue lies in that the device we are trying to connect with Cp2102, on the type-c port, the CC pin was not connected. The CC pin acts similarly to the ID/SENSE PIN in OTG cord, where it tells the android device whether to switch to host or not. Since our pin was not connected, the Android cannot recognize it, and therefore is unable to send USB ATTACHED intent out, instead it only sends out battery changed intent. 

I guess that solves it.

closed 2024-06-07 by jactor2