mik3y/usb-serial-for-android #587: Version 3.5.1 is not accessible at Jitpack

matsuob, 2024-07-20:
My gradle build fails to download the version 3.5.1:
```
Execution failed for task ':app:checkDevelopDebugAarMetadata'.
> Could not resolve all files for configuration ':app:developDebugRuntimeClasspath'.
   > Could not find com.github.mik3y:usb-serial-for-android:3.5.1.
```

https://jitpack.io/com/github/mik3y/usb-serial-for-android/3.5.1/
contains `usb-serial-for-android-3.5.1.pom`, but 
https://jitpack.io/com/github/mik3y/usb-serial-for-android/3.5.1/usb-serial-for-android-3.5.1.pom
is not found.

I think this is similar to #534.
Would it be possible for you to fix this?

kai-morich, 2024-07-20:
I can access the mentioned file. It's available again.

closed 2024-07-20 by kai-morich

matsuob, 2024-07-20:
I just confirmed that I can access it!
Thank you for your assistance.