mik3y/usb-serial-for-android #589: usbSerialPort.open(usbConnection) return connection failed：length：0；index：0

skymoomy, 2024-08-08:
how to fix this problem?

kai-morich, 2024-08-08:
Please provide more context

skymoomy, 2024-08-08:
I'm using a usb device that can be directly connected to the internet when plugged into a mobile phone。This device uses a chip developed by our company。 After download project usb-serial-for-android-master from github,i made these changes:
1. add "<uses-permission android:name="android.permission.USB_PERMISSION" />
    <uses-feature android:name="android.hardware.usb.host" android:required="true"/>" to usbSerialExamples\src\main\AndroidManifest.xml
2.add "<usb-device vendor-id="11409" product-id="2" />" to usbSerialExamples\src\main\res\xml\device_filter.xml

Install apk to my phone,i can see tree ports  in app :
USB Devices
CdsAcm,Port 0
vendor 2C91,Product 0002
CdsAcm,Port 1
vendor 2C91,Product 0002
CdsAcm,Port 2
vendor 2C91,Product 0002

But,none of them can be connected,when each of them is Clicked,it shows"connection failed:length0;index=0;"

Is there anything else that needs to be modified？Thanks

skymoomy, 2024-08-08:
> Please provide more context

I'm using a usb device that can be directly connected to the internet when plugged into a mobile phone。This device uses a chip developed by our company。 After download project usb-serial-for-android-master from github,i made these changes:

add "
" to usbSerialExamples\src\main\AndroidManifest.xml
2.add "" to usbSerialExamples\src\main\res\xml\device_filter.xml
Install apk to my phone,i can see tree ports in app :
USB Devices
CdsAcm,Port 0
vendor 2C91,Product 0002
CdsAcm,Port 1
vendor 2C91,Product 0002
CdsAcm,Port 2
vendor 2C91,Product 0002

But,none of them can be connected,when each of them is Clicked,it shows"connection failed:length0;index=0;"

Is there anything else that needs to be modified？Thanks

kai-morich, 2024-08-08:
does it work with [Serial USB Terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal) app?

skymoomy, 2024-08-09:
> does it work with [Serial USB Terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal) app?

no,it doesn't work with  [Serial USB Terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal) app,the result is same with usb-serial-for-android-master. It also shows"connection failed:length0;index=0;"

Is the driver not compatible with my usb device or some additional configuration  are requied ?

kai-morich, 2024-08-09:
your device probably is not a fully CDC ACM compliant device.

I am not aware of any place in the library that throws an exception containing something like "length0;index=0;". Is this your own exception?
With original exception, callstack and logcat output you might see where the CDC ACM setup fails.

skymoomy, 2024-08-10:
> your device probably is not a fully CDC ACM compliant device.
> 
> I am not aware of any place in the library that throws an exception containing something like "length0;index=0;". Is this your own exception?  --- "connection failed: length0;index=0; "  An exception occured when usbSerialPort.open(usbConnection) is called:

          @Override
          public void onSerialConnectError(Exception e) {
              status("connection failed: " + e.getMessage());
              disconnect();
          }

With original exception, callstack and logcat output you might see where the CDC ACM setup fails.  ---I'll try this

skymoomy, 2024-08-12:
> your device probably is not a fully CDC ACM compliant device.
>
> I am not aware of any place in the library that throws an exception containing something like "length0;index=0;". Is this your own [...]

After add log print and use "adb shell logcat" , I have identified the cause of the problem, but I do not know how to solve it:

        private void openInterface() throws IOException {

            mControlInterface = null;
            mDataInterface = null;
            int j = getInterfaceIdFromDescriptors();
            Log.d(TAG, "interface count=" + mDevice.getInterfaceCount() + ", IAD=" + j);
            Log.d(TAG, "j=" + j);
            if (j >= 0) {
                for (int i = 0; i < mDevice.getInterfaceCount(); i++) {
                    UsbInterface usbInterface = mDevice.getInterface(i);
                    Log.d(TAG, "usbInterface.i：" + i);
                    Log.d(TAG, "usbInterface.getName：" + usbInterface.getName());
                    Log.d(TAG, "usbInterface.getId：" + usbInterface.getId());
                    Log.d(TAG, "usbInterface.getInterfaceClass：" + usbInterface.getInterfaceClass());
                    Log.d(TAG, "usbInterface.getInterfaceSubclass：" + usbInterface.getInterfaceSubclass());
                    Log.d(TAG, "usbInterface.toString：" + usbInterface.toString());
                    Log.d(TAG, "usbInterface.getInterfaceProtocol：" + usbInterface.getInterfaceProtocol());
                    Log.d(TAG, "usbInterface.describeContents：" + usbInterface.describeContents());
                    if (usbInterface.getId() == j || usbInterface.getId() == j+1) {
                        if (usbInterface.getInterfaceClass() == UsbConstants.USB_CLASS_COMM &&
                                usbInterface.getInterfaceSubclass() == USB_SUBCLASS_ACM) {
                            Log.d(TAG, "mControlIndex：" + i);
                            mControlIndex = usbInterface.getId();
                            mControlInterface = usbInterface;
                        }
                        if (usbInterface.getInterfaceClass() == UsbConstants.USB_CLASS_CDC_DATA) {
                            mDataInterface = usbInterface;
                        }
                    }
                }
            }

j=2, UsbConstants.USB_CLASS_COMM = 2, USB_SUBCLASS_ACM=2,mControlIndex = 3. mControlInterface is selected as shown below, you can see that it has no mEndpoints,  "mControlEndpoint = mControlInterface.getEndpoint(0);" cause crash. 

CdcAcmSerialDriver: usbInterface.i：3
CdcAcmSerialDriver: usbInterface.getName：null
CdcAcmSerialDriver: usbInterface.getId：2
CdcAcmSerialDriver: usbInterface.getInterfaceClass：2
CdcAcmSerialDriver: usbInterface.getInterfaceSubclass：2
CdcAcmSerialDriver: usbInterface.toString：UsbInterface[mId=2,mAlternateSetting=0,mName=null,mClass=2,mSubclass=2,mProtocol=1,mEndpoints=[]
CdcAcmSerialDriver: usbInterface.getInterfaceProtocol：1
CdcAcmSerialDriver: usbInterface.describeContents：0

----------------------------------------------------------------------------------------------------------------------------------------
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.i：0
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getName：Mobile ECM Network Adapter
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getId：0
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getInterfaceClass：2
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getInterfaceSubclass：6
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.toString：UsbInterface[mId=0,mAlternateSetting=0,mName=Mobile ECM Network Adapter,mClass=2,mSubclass=6,mProtocol=0,mEndpoints=[
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: UsbEndpoint[mAddress=130,mAttributes=3,mMaxPacketSize=16,mInterval=16]]
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getInterfaceProtocol：0
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.describeContents：0
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.i：1
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getName：null
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getId：1
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getInterfaceClass：10
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getInterfaceSubclass：0
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.toString：UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[]
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getInterfaceProtocol：0
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.describeContents：0
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.i：2
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getName：null
08-12 14:22:11.657  2337  2337 D CdcAcmSerialDriver: usbInterface.getId：1
[... 37 lines cut ...]
08-12 14:22:11.658  2337  2337 D CdcAcmSerialDriver: usbInterface.getInterfaceClass：10
08-12 14:22:11.658  2337  2337 D CdcAcmSerialDriver: usbInterface.getInterfaceSubclass：0
08-12 14:22:11.658  2337  2337 D CdcAcmSerialDriver: usbInterface.toString：UsbInterface[mId=5,mAlternateSetting=0,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
08-12 14:22:11.658  2337  2337 D CdcAcmSerialDriver: UsbEndpoint[mAddress=132,mAttributes=2,mMaxPacketSize=512,mInterval=0]
[... cut ...]
08-12 14:22:11.658  2337  2337 D CdcAcmSerialDriver: usbInterface.toString：UsbInterface[mId=7,mAlternateSetting=0,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
08-12 14:22:11.658  2337  2337 D CdcAcmSerialDriver: UsbEndpoint[mAddress=133,mAttributes=2,mMaxPacketSize=512,mInterval=0]
08-12 14:22:11.658  2337  2337 D CdcAcmSerialDriver: UsbEndpoint[mAddress=5,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
08-12 14:22:11.658  2337  2337 D CdcAcmSerialDriver: usbInterface.getInterfaceProtocol：0
08-12 14:22:11.658  2337  2337 D CdcAcmSerialDriver: usbInterface.describeContents：0

kai-morich, 2024-08-12:
please provide the standard logcat output from CdcAcmSerialDriver

skymoomy, 2024-08-13:
> please provide the standard logcat output from CdcAcmSerialDriver

[logcat.zip](https://github.com/user-attachments/files/16593313/logcat.zip)

> please provide the standard logcat output from CdcAcmSerialDriver

kai-morich, 2024-08-13:
replaced logcat.zip with relevant content:
```
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: interfaces:
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbInterface[mId=0,mAlternateSetting=0,mName=Mobile ECM Network Adapter,mClass=2,mSubclass=6,mProtocol=0,mEndpoints=[
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbEndpoint[mAddress=130,mAttributes=3,mMaxPacketSize=16,mInterval=16]]
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[]
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbInterface[mId=1,mAlternateSetting=1,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=512,mInterval=0]
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbInterface[mId=2,mAlternateSetting=0,mName=null,mClass=2,mSubclass=2,mProtocol=1,mEndpoints=[]
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbInterface[mId=3,mAlternateSetting=0,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbEndpoint[mAddress=131,mAttributes=2,mMaxPacketSize=512,mInterval=0]
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbEndpoint[mAddress=3,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbInterface[mId=4,mAlternateSetting=0,mName=null,mClass=2,mSubclass=2,mProtocol=1,mEndpoints=[]
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbInterface[mId=5,mAlternateSetting=0,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbEndpoint[mAddress=132,mAttributes=2,mMaxPacketSize=512,mInterval=0]
08-13 09:20:57.103 26704 26704 D CdcAcmSerialDriver: UsbEndpoint[mAddress=4,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: UsbInterface[mId=6,mAlternateSetting=0,mName=null,mClass=2,mSubclass=2,mProtocol=1,mEndpoints=[]
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: UsbInterface[mId=7,mAlternateSetting=0,mName=null,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: UsbEndpoint[mAddress=133,mAttributes=2,mMaxPacketSize=512,mInterval=0]
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: UsbEndpoint[mAddress=5,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: trying default interface logic
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: USB descriptor:
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 12 01 00 02 EF 02 01 40 91 2C 02 00 02 00 01 02 03 01
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 09 02 09 01 08 01 00 C0 FA
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 08 0B 00 02 02 06 00 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 09 04 00 00 01 02 06 00 05
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 00 10 01
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 0D 24 0F 04 00 00 00 00 EA 05 00 00 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 06 00 01
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 07 05 82 03 10 00 10
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 09 04 01 00 00 0A 00 00 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 09 04 01 01 02 0A 00 00 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 07 05 81 02 00 02 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 07 05 01 02 00 02 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 08 0B 02 02 02 02 01 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 09 04 02 00 00 02 02 01 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 00 10 01
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 01 00 03
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 04 24 02 02
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 06 02 03
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 09 04 03 00 02 0A 00 00 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 07 05 83 02 00 02 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 07 05 03 02 00 02 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 08 0B 04 02 02 02 01 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 09 04 04 00 00 02 02 01 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 00 10 01
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 01 00 05
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 04 24 02 02
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 06 04 05
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 09 04 05 00 02 0A 00 00 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 07 05 84 02 00 02 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 07 05 04 02 00 02 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 08 0B 06 02 02 02 01 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 09 04 06 00 00 02 02 01 00
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 00 10 01
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 01 00 07
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 04 24 02 02
08-13 09:20:57.104 26704 26704 D CdcAcmSerialDriver: 05 24 06 06 07
[... cut ...]
08-13 09:20:57.105 26704 26704 D CdcAcmSerialDriver: UsbEndpoint[mAddress=5,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
08-13 09:20:57.105 26704 26704 D CdcAcmSerialDriver: usbInterface.getInterfaceProtocol：0
08-13 09:20:57.105 26704 26704 D CdcAcmSerialDriver: usbInterface.describeContents：0
08-13 09:20:57.105 26704 26704 D CdcAcmSerialDriver: usbInterface.getEndpoint(0).getType()：2
08-13 09:20:57.105 26704 26704 D CdcAcmSerialDriver: Control interface id 2
08-13 09:20:57.106 26704 26704 D CdcAcmSerialDriver: ControlInterface.getEndpointCount 0
```

kai-morich, 2024-08-13:
The `getInterfaceIdFromDescriptors` has detected the correct interface id 2+3 from IAD.
I have never seen an _Interface_ without _Endpoints_. On stackoverflow someone mentioned that _endpoint #0 must always exist and doesn't have to be mentioned in the endpoint descriptors_, so your device might be valid USB.

Now I understood the error message. 
As only ``mControlIndex`` is needed for the actual communication, does it work if you comment out the ``mControlEndpoint`` assignment and check?

skymoomy, 2024-08-14:
> The `getInterfaceIdFromDescriptors` has detected the correct interface id 2+3 from IAD. I have never seen an _Interface_ without _Endpoints_. On stackoverflow someone mentioned that _endpoint #0 [...]

After comment out the `mControlEndpoint` assignment, the port can now be successfully connected.Thks for you advice.

closed 2024-08-14 by skymoomy

kai-morich, 2024-08-16:
Looks like your device would not be accepted by Linux, see https://github.com/torvalds/linux/blob/master/drivers%2Fusb%2Fclass%2Fcdc-acm.c#L1322