mik3y/usb-serial-for-android #590: When minimizing the APP,  a crash occurred

skymoomy, 2024-08-19:
HI everyone，I encountered  problem.when the port  is successfully connected, a large amount of data will be received from the usb device.After that, if I switch to the main interface of the mobile phone, which means minimize the app (not closing it), a crash occured.Has anyone encountered the same problem before? The attachment is a log captured through logcat:

--------- beginning of crash
08-19 09:59:49.872 31569 31569 E AndroidRuntime: FATAL EXCEPTION: main
08-19 09:59:49.872 31569 31569 E AndroidRuntime: Process: com.hoho.android.usbserial.examples, PID: 31569
08-19 09:59:49.872 31569 31569 E AndroidRuntime: java.lang.RuntimeException: android.os.TransactionTooLargeException: data parcel size 626300 bytes
08-19 09:59:49.872 31569 31569 E AndroidRuntime: Bundle stats:
08-19 09:59:49.872 31569 31569 E AndroidRuntime:   androidx.lifecycle.BundlableSavedStateRegistry.key [size=624984]
08-19 09:59:49.872 31569 31569 E AndroidRuntime:     android:support:activity-result [size=1680]
08-19 09:59:49.872 31569 31569 E AndroidRuntime:       KEY_COMPONENT_ACTIVITY_REGISTERED_KEYS [size=1236]
08-19 09:59:49.872 31569 31569 E AndroidRuntime:     android:support:fragments [size=622992]
08-19 09:59:49.872 31569 31569 E AndroidRuntime:       android:support:fragments [size=622920]
08-19 09:59:49.872 31569 31569 E AndroidRuntime: PersistableBundle stats:
08-19 09:59:49.872 31569 31569 E AndroidRuntime:   [null]
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.app.servertransaction.PendingTransactionActions$StopInfo.run(PendingTransactionActions.java:146)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:958)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:224)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:318)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:8711)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:561)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1013)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: Caused by: android.os.TransactionTooLargeException: data parcel size 626300 bytes
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.os.BinderProxy.transactNative(Native Method)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.os.BinderProxy.transact(BinderProxy.java:621)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.app.IActivityClientController$Stub$Proxy.activityStopped(IActivityClientController.java:1459)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.app.ActivityClient.activityStopped(ActivityClient.java:112)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	at android.app.servertransaction.PendingTransactionActions$StopInfo.run(PendingTransactionActions.java:135)
08-19 09:59:49.872 31569 31569 E AndroidRuntime: 	... 8 more

[logcat095949.zip](https://github.com/user-attachments/files/16654568/logcat095949.zip)

kai-morich, 2024-08-19:
When an App is send to background, rotated, ... the state is saved by Android automatically. This includes the text view content and fails if the data structure is larger than ~ 1MB. Workaround is saving the fragment state manually.

skymoomy, 2024-08-28:
fix

skymoomy, 2024-08-28:
fix

closed 2024-08-28 by skymoomy