mik3y/usb-serial-for-android #593: how to get multiple interface info?

skymoomy, 2024-08-30:
Hi,all:
My usb device has three ports on PC：
USB\VID_2C91&PID_0002&MI_06\6&3911747D&1&0006
USB\VID_2C91&PID_0002&MI_02\6&3911747D&1&0002
USB\VID_2C91&PID_0002&MI_04\6&3911747D&1&0004

also, it has three ports on my phone,but i can only get port num in the port list, which comes from these codes:
                    text1.setText(item.driver.getClass().getSimpleName().replace("SerialDriver","")+", Port "+item.port);

I know how to get VID&PID, but i don't kown how to get MI ?
I need to obtain MI information and distinguish these ports

kai-morich, 2024-08-30:
I never heard of MI values, but very likely your device returns the ports in a fixed order so you can try out

skymoomy, 2024-08-31:
https://learn.microsoft.com/en-us/windows-hardware/drivers/install/standard-usb-identifiers

There is some information about multiple interfaces on this website：

Multiple-Interface USB Devices
Devices with multiple interfaces are called composite devices. Starting with Windows 2000, when a new [USB composite device](https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/register-a-composite-driver) is plugged into a computer, the USB hub driver creates a physical device object (PDO) and notifies the operating system that its set of child devices has changed. After querying the hub driver for the hardware identifiers associated with the new PDO, the operating system searches the appropriate INF files to find a match for the identifiers. If it finds a match other than USB\COMPOSITE, it loads the driver indicated in the INF file. However, if no other match is found, the operating system uses the compatible ID USB\COMPOSITE, for which it loads the USB Generic Parent driver. The Generic Parent driver then creates a separate PDO and generates a separate set of hardware identifiers for each interface of the composite device.

Each interface has a device ID of the following form:

USB\VID_v(4)&PID_d(4)&MI_z(2)

Where:

v(4) is the vendor code that the [USB committee assigns to the vendor](https://www.usb.org/developers).

d(4) is the product code that the vendor assigns to the device.

z(2) is the interface number that is extracted from the bInterfaceNumber field of the corresponding [interface descriptor](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/usbspec/ns-usbspec-_usb_interface_descriptor).

kai-morich, 2024-09-01:
the ports are numbers in the order as they are returned by the device. This typically is fixed. Please try and use 0,1,2 instead of 2,4,6

closed 2025-06-16 by kai-morich