mik3y/usb-serial-for-android #599: Missing/Corrupt data with high baud rates.

GeorgePapageorgakis, 2024-10-07:
Hello,

I have a serial device that sends data with the following configuration using an `FTDI FT230XQ-R` chip:

```
baud rate = 921600
databits = 8
stopbit = 1
parity = none
dtr = true
rts = true
Flow control = false
```

The data are confirmed to be send correctly in some other serial interfaces that was tested with same specs.
However, in Android, we noticed that we receive corrupted data at this baud rate. 
At a lower baud rate, such as 112500, the data corruption seems to be rarer but still occurs at fewer bytes.
At the high baud rate of 921600, when processing chunks of data, we noticed that last 62 bytes of the chunk were zeros.

Now, we noticed that this issue also occurs in your Android app, `Simple Usb Terminal` and the official app `Serial USB Terminal`.
From our side, for now we cannot have Flow Control due to firmware limitations.

What are your thoughts on this?
Is it possible that the usb converter could be the cause of this issue?
Its a common `TpLink usb-c to usb 3.0 adapter UC400(UN) 1.0`

Thanks in advance.

GeorgePapageorgakis, 2024-10-07:
Just tested by adding 1 to 10ms every 1KB. The problem persists, still getting the last 62 bytes as zeros of every chunk.
For example, with default `4096` as buffer in `SerialInputOutputManager` data is split in chunks, say:
`3968 bytes x8` chunks and in every chunk the last 62 bytes are zeroes. 

Thus, the larger and more chunked the initial packet the more corrupt the data.

kai-morich, 2024-10-07:
you could try to tweak thread priority as mentioned [here](https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#i-observed-read-data-loss-with-continuous-transfer-at-high-baud-rates) but preventing data loss is hardly possible without flow control.

Each FTDI request begins with a 2 byte header so it only has 62 byte of actual content. If the 62 bytes are fully used is up to the FTDI device.

You might try FT2232 or FT4232 as these have larger buffers.

GeorgePapageorgakis, 2024-10-08:
@kai-morich Ok so Flow Control is mandatory for these high baud rates.

A workaround that seems to work is to send chunks of max size with a pause. Now it seems to work with 1KB chunk and 50ms delay. However, this is also non deterministic approach. It improved reliability but now if we have larger packets, say 100-150KB data, only 64KB seem to pass through, rest are lost/corrupted. Needs more testing/debugging to understand what is going on now.

chaniotisalex, 2024-10-11:
Thanks @GeorgePapageorgakis . Chunking solved the issue in my case too.

GeorgePapageorgakis, 2024-10-13:
I would recommend to add an additional overloaded constructor in UsbInputOutputManager with buffer size parameter. The default of 4KB seems problematic and increasing it to 16KB or more, improved the performance and chunking issues.

kai-morich, 2024-10-14:
you can already configure the buffer sizes with `setReadBufferSize` and `setWriteBufferSize` in `SerialInputOutputManager` class

jhlink, 2025-02-27:
@GeorgePapageorgakis @chaniotisalex @kai-morich 

I was experiencing something similar to you with the CH34X driver for incoming reads with a 3MB payload.

I just reached an RCA yesterday related to this and intend to submit a PR for the update after agreeing with @kai-morich on options.

Long story short,  in the SerialInputOutputManager class, the ReadBuffer capacity is by default set to the max packet size of the bulk read endpoint. Typically, this ranges from 8-64 bytes depending on hardware. Depending on how the serial driver for a given hardware was implemented, the **packet size of sent packets ** is somewhere between (0, maxPacketSizeOfEndpoint] bytes as seen in bulk transfer requests. 

In other words, if I'm sending a 2KB chunk bulk transfer request, the payload is broken down into packets with a size equal to the endpoint's max packet size or maybe 1-3 bytes on the USB protocol layer. (Why the variant packet size? Ask the vendor.) Thus, eventually buffer overflow occurs as the read buffer saturates at a fraction of the 2KB payload in the request. 

While I had a generous buffer capacity when handling onNewData byte arrays, the buffer overflow was occurring in the ReadBuffer within  SerialOutputInputManager. I was experiencing data loss and corruption around 20-30% of a serial payload transfer due to buffer overflow. After updating this internal readBuffer size to 4kb, I no longer saw data loss and corruption.

My suggestion would be to update the readBuffer capacity within SerialInputOutputManager to be at least equal to or greater than the write buffer size. 

https://github.com/mik3y/usb-serial-for-android/blob/026355f61e8d99ad7e9f639d8e4eb81fbb323c55/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/util/SerialInputOutputManager.java#L44-L74

Or if people don't mind spending 16KB, setting the read buffer capacity equal to the old bulk transfer limit of 16KB per the logic in CommonUsbSerialPort.

https://github.com/mik3y/usb-serial-for-android/blob/026355f61e8d99ad7e9f639d8e4eb81fbb323c55/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/CommonUsbSerialPort.java#L32

Any thoughts @kai-morich on the suggestion?

kai-morich, 2025-02-28:
Increasing the readbuffer by default has negative impact if data is coming in steadily at not so high rate. I observed cases where the call did not return before the buffer was full, causing lagging updates in terminal Apps.

Please try [readqueue](https://github.com/mik3y/usb-serial-for-android/tree/readqueue) branch with new `readQueue` method at `SerialInputOutputManager/UsbSerialPort` classes. The new feature is enabled by default in SerialInputOutputManager with READ_QUEUE_BUFFER_COUNT = 4;
If it solves your issue, please let me know the `bufferCount` that worked for you.

kai-morich, 2025-02-28:
some devices fill up the packets, others don't. For those that fill up, a larger buffersize caused lagging as the request did not return until the larger buffer was full. At low baud rate was already in the range of seconds.

jhlink, 2025-03-01:
@kai-morich 
So, just a couple updates. Initially, setting the read and write buffer size to 4kb in SerialInputOutputManager seems to work for the CH34x, but not for the MCP2200. After sending the second 3 MB payload, I experience data loss/corruption. 

I reset the custom buffer size in SerialInputOutManager and used the default settings for the readqueue branch. 
Worked fine over 5 tests where I sent a 3 MB payload with the MCP2200. 
On the CH34X, the first attempt succeeded. On the second attempt, I experienced data loss. 

I'll be experimenting with what readQueue count works for both chips over the next few days. 
The goal is passing 5 attempts sending 3 MB, each. 
Trying 8 for the CH34X, right now. 

Thanks for the work on the readqueue branch!

jhlink, 2025-03-04:
@kai-morich 

Alrighty. 8 Read Buffers was not enough. Data loss occurred on the CH34X on the 4th attempt. 

32 Read Buffers is working pretty well so far. 

With a 3 MB payload...
7 consecutive transfers on the MCP2200
11 consecutive transfer on the CH34X.

kai-morich, 2025-03-04:
What is `port.getReadEndpoint().getMaxPacketSize()` for your devices? I assume 32 or 64 so the total queue size is not more than 4kb?

kai-morich, 2025-03-07:
Which baud rate did you use?

jhlink, 2025-03-07:
@kai-morich 

For both devices, 
115200
8 Data Bits
1 Stop Bit
No Parity

kai-morich, 2025-03-09:
115k is not very high. Do you do significant processing inside `onNewData`?

jhlink, 2025-03-10:
@kai-morich . I agree. 

Depends what qualifies as significant. 

For every new event payload in onNewData, I'm iterating through incoming byte array for a terminating byte and appending it to a byte buffer.  
On discovering a terminating byte, I allocate a new byte array equal to the size of the incoming payload, copy src bytes to that byte array, then queue the data as a Message to the Main Handler thread for processing.

Hameedoooooo, 2025-05-10:
> Hello,
> 
> I have a serial device that sends data with the following configuration using an `FTDI FT230XQ-R` chip:
> 
> ```
> baud rate = 921600
> databits = 8
> stopbit = 1
> parity = none
> dtr = true
> rts = true
> Flow control = false
> ```
> 
> The device functions correctly on other serial interfaces with identical configurations, indicating the data is transmitted accurately. However, when connected to Android devices, we observe corrupted data particularly at the higher baud rate of `921600`. At this rate, the last `62` bytes of each data chunk consistently register as zeros. Lowering the baud rate to `112500` decreases the frequency of these errors, though they still persist on occasion.
> 
> This problem has also been replicated in the Simple USB Terminal and Serial USB Terminal apps on Android. Unfortunately, enabling flow control is not feasible due to our current firmware constraints.
> 
> Could the USB converter, specifically the `TpLink USB-C to USB 3.0 adapter UC400(UN) 1.0`, be contributing to this issue? 
> We would greatly appreciate any insights or suggestions you might have to help resolve this data corruption.
> 
> Could this be the solution? https://github.com/mik3y/usb-serial-for-android/wiki/FAQ#when-reading-a-message-why-dont-all-bytes-arrive-at-the-same-time
> 
> So for 512 bytes a 1-10ms pause could work? The size of the whole packet could reach up to 1-2MB.
> 
> Thank you for your assistance.