mik3y/usb-serial-for-android #600: Publish artifacts to Maven Central

erawhctim, 2024-10-14:
Any chance you could support this? 

jitpack is convenient for library authors but inconvenient for consumers (especially those at large organizations with requirements about artifact hosting).

neboskreb, 2024-10-19:
@kai-morich 
I can provide support, if you like. 

It will take some emails on your end to register you repo with Maven Central; once that done, I can do the rest of lifting. Let me know.

Jai-GAY, 2024-10-28:
does anyone face [the same issue](https://github.com/mik3y/usb-serial-for-android?tab=readme-ov-file#quick-start)?
```
allprojects {
    repositories {
        ...
        maven { url 'https://jitpack.io' }
    }
}
```
Could not GET 'https://jitpack.io/com/github/supervital/swipenumberpicker/1.0.3/swipenumberpicker-1.0.3.pom'. Received status code 401 from server: Unauthorized

erawhctim, 2024-10-28:
@kai-morich @mik3y Does migrating to Maven Central seem feasible, or is there some blocker here?

kai-morich, 2024-10-31:
I signed up at sonatype and played around with registering namespaces.
This library uses `package com.hoho.android.usbserial` for the Java classes.
Do namespaces and package name have to be identical?

As I am not the original author of this library I do not have DNS write access to hoho.com and no full write access to this project, e.g. I cannot store github secrets in this project which are very likely needed for automated integration. Might need involvement of the original author @mik3y

neboskreb, 2024-10-31:
In this case the Java package name can stay `com.hoho` but the coordinate of the artifact has to change to `io.github.mik3y`. 

If you want automatic publishing of the releases then - yes, it will require installing secrets.

Having only one admin user looks to me a big risk; delegating the admin permissions to at least one more person might be a good idea. What do you think, @mik3y ?