mik3y/usb-serial-for-android #603: Using Native C library with USB File descriptor FAILS for write and read

BVmamatha, 2024-10-25:
Hi,
We are using Usb-Serial-for-android library in our android project. We have integrated our C  library into android project which communicates with USB port. Device has 2 ports connected. We are able to get the permission for these ports and able to read and write in MainActivity. But we need to pass the filedescriptor (Connection.getFileDescriptot()) to our C library which handles reading and writing to these ports. The C library then calls the callback functions which are defined in Android app .
Problem : The filedescriptor passed to C lib, fails to write and read.  Please check this issue at the earliest.
[usbnew.zip](https://github.com/user-attachments/files/17515896/usbnew.zip)

BVmamatha, 2024-10-25:
write operation in C library return EINVAL (invalid filedescriptor) error.

kai-morich, 2024-10-25:
please check https://github.com/mik3y/usb-serial-for-android/wiki/Projects-Using-usb-serial-for-android for wrappers to other languages, but I doubt anyone has used C yet

closed 2024-10-25 by kai-morich