mik3y/usb-serial-for-android #605: Could not find com.github.mik3y:usb-serial-for-android:3.8.1

Jai-GAY, 2024-10-28:
com.github.mik3y:usb-serial-for-android:**3.8.0** was okay.

```
buildscript {
    repositories {
        mavenCentral()
        google()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:8.4.2'
    }
}
allprojects {
    repositories {
        mavenCentral()
        maven { url 'https://jitpack.io' }
        google()
     }
}
> Could not find com.github.mik3y:usb-serial-for-android:3.8.1.
     Searched in the following locations:
       - https://repo.maven.apache.org/maven2/com/github/mik3y/usb-serial-for-android/3.8.1/usb-serial-for-android-3.8.1.pom
       - https://jitpack.io/com/github/mik3y/usb-serial-for-android/3.8.1/usb-serial-for-android-3.8.1.pom
       - https://dl.google.com/dl/android/maven2/com/github/mik3y/usb-serial-for-android/3.8.1/usb-serial-for-android-3.8.1.pom
     Required by:
         project :app > project :XXXXXXXX
```

kai-morich, 2024-10-28:
3.8.1 is not released yet, but already mentioned on overview page. was changed as preparation for upcomming release

kai-morich, 2024-10-28:
now it's available

closed 2024-10-28 by kai-morich

bentangyy, 2024-12-25:
Could not find com.github.mik3y:usb-serial-for-android:3.8.1.
Required by:
    project :app