mik3y/usb-serial-for-android #611: AndrOBD: Null pointer exception on connecting a USB device

fr3ts0n, 2024-12-15:
Hi, I have a problem in AndrOBD causing a NullPointerException when connecting the selected USB device:
Used versions of lib: V3.8.1, V3.8.0
Last known good version: V3.3.0

It appears to happen within the instantiation of the SerialInputOutputManager.

Here is the call stack:
`java.lang.RuntimeException: Failure delivering result ResultInfo{who=null, request=6, result=-1, data=Intent {  }} to activity {com.fr3ts0n.ecu.gui.androbd/com.fr3ts0n.ecu.gui.androbd.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.hardware.usb.UsbEndpoint.getMaxPacketSize()' on a null object reference
	at android.app.ActivityThread.deliverResults(ActivityThread.java:5339)
	at android.app.ActivityThread.handleSendResult(ActivityThread.java:5378)
	at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:67)
	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:138)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2307)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7924)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.hardware.usb.UsbEndpoint.getMaxPacketSize()' on a null object reference
	at com.hoho.android.usbserial.util.SerialInputOutputManager.<init>(SerialInputOutputManager.java:73)
	at com.fr3ts0n.ecu.gui.androbd.UsbCommService.setDevice(UsbCommService.java:120)
	at com.fr3ts0n.ecu.gui.androbd.UsbCommService.connect(UsbCommService.java:127)
	at com.fr3ts0n.ecu.gui.androbd.MainActivity.onActivityResult(MainActivity.java:967)
	at android.app.Activity.dispatchActivityResult(Activity.java:8665)
	at android.app.ActivityThread.deliverResults(ActivityThread.java:5332)
	at android.app.ActivityThread.handleSendResult(ActivityThread.java:5378) 
	at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:67) 
	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45) 
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:138) 
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95) 
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2307) 
	at android.os.Handler.dispatchMessage(Handler.java:106) 
	at android.os.Looper.loopOnce(Looper.java:201) 
	at android.os.Looper.loop(Looper.java:288) 
	at android.app.ActivityThread.main(ActivityThread.java:7924) 
	at java.lang.reflect.Method.invoke(Native Method) 
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548) 
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936) 
`

Do you have any idea if I have to change comething in the initialisation code from V3.3.0 to V3.8.0?

Thank you very much

kai-morich, 2025-01-04:
as of v3.3.2 the mReadBuffer is initialized in the SerialInputOutputManager constructor. This only works if the UsbSerialPort is already opened. I will check if the initialization can be moved to a later function.

fr3ts0n, 2025-01-07:
THX. I have changed the sequence. Seems to work. :))

kai-morich, 2025-01-07:
https://github.com/fr3ts0n/AndrOBD/commit/8750c9b017ca082dbc4a37c3a1e91d4f9587ba6b

closed 2025-06-16 by kai-morich