mik3y/usb-serial-for-android #617: Can This library be used in Processing for android IDE?

jhsa, 2025-02-12:
As the title say, Is it possible to use this Library in Processing IDE in android Mode?
Thanks.

kai-morich, 2025-02-13:
I don't know. Please check the wiki https://github.com/mik3y/usb-serial-for-android/wiki/Projects-Using-usb-serial-for-android

jhsa, 2025-02-13:
> I don't know. Please check the wiki https://github.com/mik3y/usb-serial-for-android/wiki/Projects-Using-usb-serial-for-android

  I did that already before I commented but didn't find anything, hence my question.

kai-morich, 2025-02-14:
I found some stackoverflow entries that mention that you can add .jar files.
Check [here](https://github.com/mik3y/usb-serial-for-android/wiki/Build-Variants#jar) how to build a .jar file for this library

closed 2025-02-14 by kai-morich