mik3y/usb-serial-for-android #619: Does this library not support 921600 baud rate?

Game106, 2025-03-07:
**Software:** 
    android studioAndroid Studio Electric Eel | 2022.1.1 Patch 2

**Test phone:** Huawei Honor X10
**System:** HarmonyOS 2.0.0

**Problem description：** 
    I use this library to receive serial data with a baud rate of 921600, but I always can't get the complete data. The microcontroller sends 1024 bytes of data at a baud rate of 921600 and only receives 800-900 bytes of data. Sending at 115200 baud rate can receive 1024 bytes of data normally. 
    Does this library not support 921600 baud rate? 
    What should I do to receive complete data at 921600 baud rate? 
    
Thank you

kai-morich, 2025-04-06:
please try the _readqueue_ branch. This includes optimizations for high baud rates

Game106, 2025-04-06:
来信我已收到，谢谢。