mik3y/usb-serial-for-android #625: net9.0-android binding library compatibility

dormakabahar, 2025-05-21:
Using this library in dotnet9-android project created with Visual Studio 2022.
Android binding library project builds only successfull with versions of this library up to 3.7.3.

With more recent versions (3.8.0 and upwards) of this library, there are errors when building the dotnet android library binding project:

Error (active)	CS0535	'CommonUsbSerialPort' does not implement interface member 'IUsbSerialPort.GetFlowControl()'	AndroidBinding1	E:\TEMP\usb-serial-for-android__TEST\AndroidBinding1\AndroidBinding1\obj\Debug\net9.0-android\generated\src\Com.Hoho.Android.Usbserial.Driver.CommonUsbSerialPort.cs	20	

Severity	Code	Description	Project	File	Line	Suppression State
Error (active)	CS0535	'CommonUsbSerialPort' does not implement interface member 'IUsbSerialPort.SetFlowControl(IUsbSerialPort.FlowControl?)'	AndroidBinding1	E:\TEMP\usb-serial-for-android__TEST\AndroidBinding1\AndroidBinding1\obj\Debug\net9.0-android\generated\src\Com.Hoho.Android.Usbserial.Driver.CommonUsbSerialPort.cs	20	

It would be good if this library can also be used out of the box with newest version in Visual Studio Android projects.

[image]

kai-morich, 2025-05-21:
with 3.8.0 the UsbSerialPort interface got more methods, as you can check [here](https://github.com/mik3y/usb-serial-for-android/commit/88ca3f57c423941b4a918e91a7231449512fa1e0#diff-0083a3832998738e885e4647ce23c7d5e0952a39619499c18c5d77bb5e8cc167). Probably your .net wrapper needs to be enhanced.

dormakabahar, 2025-05-26:
What do you mean with enhancing my .net wrapper? Can you give me a hint? The Screenshot shows the only code in the wrapper. Worked without any error up to 3.7.3.

kai-morich, 2025-05-26:
I am not familiar with net9.0-android binding library.