mik3y/usb-serial-for-android #627: Library file issues

Kasslim, 2025-05-21:
I am trying to build an app with this library included but am running into problems. I have tried adding jitpack.io to the repositories and it is able to GET the .pom file, but not the .aar file.

**Context & Info**
This a fresh Android Kotlin project where all I've tried to do is add this library.

Visiting `https://jitpack.io/#mik3y/usb-serial-for-android` shows "Please Sign In first".

Visiting `https://jitpack.io/com/github/mik3y/usb-serial-for-android/3.9.0/usb-serial-for-android-3.9.0.aar` with a browser yielded an empty file 2 days ago, now it gives a 56KB file (maybe because I signed into Github on my browser?). This by itself already makes me think something funny is going on. But, when manually inserting this .aar file into the project as a dependency the build still does not work and I get the following errors:

<details>
  <summary>Build failure exceptions</summary>
  
```
FAILURE: Build completed with 3 failures.
 
1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':automotive:processDebugResources'.
> Could not resolve all files for configuration ':automotive:debugRuntimeClasspath'.
> Failed to transform usb-serial-for-android-3.9.0.aar to match attributes {artifactType=android-compiled-dependencies-resources}.
> Execution failed for AarResourcesCompilerTransform: <USER>\.gradle\caches\8.10.2\transforms\05561c05ba74cf9b418382d0a81d4fea\transformed\usb-serial-for-android-3.9.0.
> <USER>\.gradle\caches\8.10.2\transforms\05561c05ba74cf9b418382d0a81d4fea\transformed\usb-serial-for-android-3.9.0\AndroidManifest.xml
 
* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.
==============================================================================
 
2: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':automotive:checkDebugDuplicateClasses'.
> Could not resolve all files for configuration ':automotive:debugRuntimeClasspath'.
> Failed to transform usb-serial-for-android-3.9.0.aar to match attributes {artifactType=enumerated-runtime-classes, org.gradle.usage=java-runtime}.
> Execution failed for AarToClassTransform: <USER>\AndroidStudioProjects\test\automotive\libs\usb-serial-for-android-3.9.0.aar.
> invalid END header (bad central directory offset)
 
* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.
==============================================================================
 
3: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':automotive:desugarDebugFileDependencies'.
> Could not resolve all files for configuration ':automotive:debugRuntimeClasspath'.
> Failed to transform usb-serial-for-android-3.9.0.aar to match attributes {artifactType=android-classes-jar, org.gradle.usage=java-runtime}.
> Execution failed for AarToClassTransform: <USER>\AndroidStudioProjects\test\automotive\libs\usb-serial-for-android-3.9.0.aar.
> invalid END header (bad central directory offset)
 
* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.
==============================================================================
 
BUILD FAILED in 1m 16s
24 actionable tasks: 4 executed, 20 up-to-date
```
  </details>

My gradle files

<details>
  <summary>Project build.gradle.kts</summary>
  
```
plugins {
    alias(libs.plugins.android.application)
    alias(libs.plugins.kotlin.android)
}

android {
    namespace = "com.example.test"
    compileSdk = 35

    defaultConfig {
        applicationId = "com.example.test"
        minSdk = 34
        targetSdk = 35
        versionCode = 1
        versionName = "1.0"

        testInstrumentationRunner = "androidx.test.runner.AndroidJUnitRunner"
    }

    buildTypes {
        release {
            isMinifyEnabled = false
            proguardFiles(
                getDefaultProguardFile("proguard-android-optimize.txt"),
                "proguard-rules.pro"
            )
        }
    }
    compileOptions {
        sourceCompatibility = JavaVersion.VERSION_11
        targetCompatibility = JavaVersion.VERSION_11
    }
    kotlinOptions {
        jvmTarget = "11"
    }
}

dependencies {

    implementation(libs.androidx.core.ktx)
    implementation(libs.androidx.appcompat)
//    implementation(files("libs/usb-serial-for-android-3.9.0.aar"))
    testImplementation(libs.junit)
    androidTestImplementation(libs.androidx.junit)
    androidTestImplementation(libs.androidx.espresso.core)
//    implementation(libs.usb.serial)
//    implementation("com.github.mik3y:usb-serial-for-android:v3.9.0")
//    implementation("com.github.mik3y:usb-serial-for-android:3.9.0")
    implementation(fileTree(mapOf("dir" to "libs", "include" to listOf("*.jar", "*.aar"))))
}
```
</details>

<details>
  <summary>settings.gradle.kts</summary>
  
```
pluginManagement {
    repositories {
        google {
            content {
                includeGroupByRegex("com\\.android.*")
                includeGroupByRegex("com\\.google.*")
                includeGroupByRegex("androidx.*")
            }
        }
        mavenCentral()
        gradlePluginPortal()
    }
}
dependencyResolutionManagement {
    repositoriesMode.set(RepositoriesMode.FAIL_ON_PROJECT_REPOS)
    repositories {
        google()
        mavenCentral()
        maven { url = uri("https://jitpack.io") }
    }
}

rootProject.name = "test"
include(":automotive")
```
</details>

The feeling I get from the errors is that the .aar file is corrupted, but that seems unlikely. What is going wrong?

kai-morich, 2025-05-21:
I downloaded the file without issues. The correct size is 64.757 bytes. Please try again, maybe you have to clear some cache to get rid of the broken file

Kasslim, 2025-05-22:
Oh very strange, the one I keep downloading is 56 565 bytes. I have tried clearing browser cache and even downloading on another computer, but that doesn't change anything.

I will try downloading it from another network and see if that makes a difference, will post an update after that.

P.S. Thanks for the quick response!

Kasslim, 2025-05-23:
Tried on another network with another computer, file is 64.757 bytes and the build/import now seems to work! Network administrators are trying to sabotage me it seems 😓 

Thanks for your help and have a great day! 😄

closed 2025-05-23 by Kasslim