mik3y/usb-serial-for-android #632: missing response

tcurdt, 2025-08-05:
When sending `?` to GRBL I should get back an immediate response.
And I can confirm this behaviour on a normal terminal.

```
?<Idle|MPos:0.000,0.000,0.000|Bf:100,1023|F:0>
```

Trying to replicate this:

```
                val command = "?"
                val data = command.toByteArray(Charsets.UTF_8)
                port.write(data, 1000)
                // port.purgeHwBuffers(true, false)
                
                val bytesHex1 = data.joinToString(" ") { "%02x".format(it) }
                Log.d(TAG, "tx[${data.size}]: [${command}] bytes:[${bytesHex1}]")

                // sleep(10)

                val buffer = ByteArray(1024)
                val numBytes = port.read(buffer, 1000)
                val newData = String(buffer, 0, numBytes)

                val bytesHex2 = buffer.take(numBytes).joinToString(" ") { "%02x".format(it) }
                Log.i(TAG, "rx[$numBytes]: [${newData}] bytes:[${bytesHex2}]")
```

It is sending - but I am not getting a response.

The communication works fine for line based communication though.

What am I missing? Any pointers how to debug this further?

I can only imagine buffering to cause a problem here.

tcurdt, 2025-08-06:
a single byte write does not seem to get flushed - at least for the `CdcAcmSerialDriver`

kai-morich, 2025-08-07:
(https://github.com/grbl/grbl/wiki/Using-Grbl) says that you have to use 'carriage return' as newline.

kai-morich, 2025-08-08:
I tried with a CDC device. single bytes are send with this library and received by the CDC device.
Probably it is an issue with CDC library usage on your particular device, or the the application expects some kind of newline as end-of-command delimiter

tcurdt, 2025-08-08:
It works just fine with a terminal on my laptop.
And it works just fine with another serial library on Android.
To me that does point to problem being `usb-serial-for-android`.

> or the the application expects some kind of newline as end-of-command delimiter

Nope

https://github.com/gnea/grbl/blob/master/doc/markdown/commands.md#ascii-realtime-command-descriptions

tcurdt, 2025-08-08:
```
port.dtr = true
```

indeed - that fixed it. Thanks for the help.

The FAQ entry heading made it felt unrelated and made me skip it.

closed 2025-08-08 by tcurdt

kai-morich, 2025-08-11:
I now mention DTR more prominently in the FAQ