mik3y/usb-serial-for-android #633: How do I switch the alternate settings?

magicbaby810, 2025-09-03:
my device need switch alternate settings. but I don't know how to do. UsbInterface have not the method to setAlternateSetting.

magicbaby810, 2025-09-03:
[image]

I need use the interface 0, Alt Setting 1. But `getInterface(0)` only get the Alt setting 0.

kai-morich, 2025-09-03:
With https://github.com/kai-morich/usb-serial-for-android-mik3y/blob/c608aadc59d2d09f5e34ab928de42c16bf69864f/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/CdcAcmSerialDriver.java#L59 and  https://github.com/kai-morich/usb-serial-for-android-mik3y/blob/c608aadc59d2d09f5e34ab928de42c16bf69864f/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/CdcAcmSerialDriver.java#L224 the library reports all interface combinations that have required comm + data interface for serial communication , so your alternative setting should be included and driver.getPorts() could return more than 1

magicbaby810, 2025-09-03:
driver.getPorts() return two ports. But then? I need change the alternate setting 0 to  1.

magicbaby810, 2025-09-03:
In the Interface, the interfaceclass is 255,  subclass is 255 , protocol is 255 . so countPorts method is return 0

kai-morich, 2025-09-03:
As your device returns vendor specific values (0xff) it does not work with the existing CDC-ACM driver in this library. 
Looks like you need a dedicated library for this particular device or would have to implement a dedicated driver that could be added to this library. 
Adding a dedicated driver to this library only makes sense if it is of general use.

magicbaby810, 2025-09-04:
> As your device returns vendor specific values (0xff) it does not work with the existing CDC-ACM driver in this library. Looks like you need a dedicated library for this particular device or would [...]

Very much thanks, You are the github master of the fastest reply who I've even seen. Although my question haven't resolved, but I still thank for you.